wled/WLED #946: only WLED_0.8.5_ESP8266.bin works on my device

dogson3006, 2020-05-25:
hi, 
my problem is, that only WLED_0.8.5_ESP8266.bin works on my device.
 if i flash for example WLED_0.10.0_ESP8266.bin, i can find the AP. but if i type in my data to local wlan,  it didnt connect correctly.
 the device will shown in the list of my router,  but when i try to connect to the displayed IP, no gui appears and the error running out of time.
plz give me a hint
thank you

SimState, 2020-05-26:
Same here! only 0.8.5 works...

Legsmaniac, 2020-05-26:
> if i flash for example WLED_0.10.0_ESP8266.bin

Instead of flashing WLED_0.10.0_ESP8266.bin, have you tried updating to WLED_0.10.0_ESP8266.bin via OTA from the UI?  Doing that should maintain your settings with no need to use AP

aryedevops, 2020-05-26:
I have had the same issue with earlier versions - and ONLY 0.8.5 works!! I was getting disconnected all the time and getting RED messages all over. I wonder why?

Aircoookie, 2020-05-26:
Hi everyone! I suspect that this issue is due to the Arduino core version, not any changes to WLED.
Here is the latest version compiled with core `2.5.2`, same as 0.8.5:
https://drive.google.com/open?id=1g-5s3WtJ_VgGq-oSdHazuABUNJBskd1i
Let me know if that works better!

dogson3006, 2020-05-26:
> 
> 
> > if i flash for example WLED_0.10.0_ESP8266.bin
> 
> Instead of flashing WLED_0.10.0_ESP8266.bin, have you tried updating to WLED_0.10.0_ESP8266.bin via OTA from the UI? Doing that should maintain your settings with no need to use AP

good idea. but i did this the first time to update to new version. after selecting the file, the update starts and a message said: update successfull, rebooting. after that i ll cant connect again. same error. 
ip is given from router but the gui will not open. allways: running out of time.

dogson3006, 2020-05-26:
> 
> 
> Hi everyone! I suspect that this issue is due to the Arduino core version, not any changes to WLED.
> Here is the latest version compiled with core `2.5.2`, same as 0.8.5:
> https://drive.google.com/open?id=1g-5s3WtJ_VgGq-oSdHazuABUNJBskd1i
> Let me know if that works better!

thanks man, you are the best. this version works perfectly.

velidhu, 2020-05-27:
i think the problem was that it didn't change wifi settings if the esp had been used for other projects. with a new esp no problem. with the new core it also works on recycled esp

closed 2020-05-28 by dogson3006

pontos-dev, 2020-06-01:
@Aircoookie It work's with that binary you linked to :)
Thx

acorn1sh, 2020-06-03:
> Hi everyone! I suspect that this issue is due to the Arduino core version, not any changes to WLED.
> Here is the latest version compiled with core `2.5.2`, same as 0.8.5:
> https://drive.google.com/open?id=1g-5s3WtJ_VgGq-oSdHazuABUNJBskd1i
> Let me know if that works better!

So I have noticed that this binary is flashing when I choose an effect. I loaded 10.0 and I have no flashing but when I use this file my lights will flash like issue #230. If I choose a solid color it seems to be fine.

jonhawkes, 2020-06-10:
I needed this to stop the web front end hanging for me. Works fine.