wled/WLED #947: LOLin Wifi Issue

Doyle4, 2020-05-26:
So I setup my Wifi Kit 8, no problems at all.

I do the same for my LOLin, flashes, I do the setup... then nothing, nothing will find the phone, seems as soon as setup and hard rest is done, its like the device has been wiped, so i reflash... same thing again, Iv noticed its when i change either .local name or wifi device name.

mike2nl, 2020-06-11:
@Doyle4 
Try to use a real IP address and NOT the .local DNS part. Dpending on your router the DNS won't work. It is a known issue with many router because wrong DNS implementation.

Doyle4, 2020-06-11:
I will give this a try.

I did however spot was going wrong. Its not the board erasing, the SSD is hidden even when unticked, not sure if its a bug or is meant to be like this.

closed 2020-10-17 by stale[bot]