wled/WLED #948: Board Recommendation

Legsmaniac, 2020-05-26:
Up to now, I have always used [NodeMCU 12-F DOITING WiFi+ ESP8266](https://www.amazon.co.uk/gp/product/B07M8Q38LK) boards with no problems whatsoever.  I only started with these boards after watching Dr Zzz's YT vid back in January which started it all off for me and this board looked to be similar to the one he was using.   No problems except for one or two which could do with better WiFi reception.  So based on Christian's recommendation (on his download page [right here](https://github.com/Aircoookie/WLED/releases)) I decided to get [WeMos D1 Mini Pro](https://www.amazon.co.uk/gp/product/B083M6N55Z) which has an antenna socket as well as being smaller and more compact than what I was currently using.   But I think I'm about to send these back for refund!

The problems?  WiFi keeps dropping/disappearing and only comes back if I reboot the board. (And yes, I've tried the WiFi usermod recently added, doesn't do anything), Randomly freezes and sometimes just switches itself off.  I even tried version 0.8.5 with exactly the same problems.  I got 4 (2 packs of 2) of these boards and they all do the same thing.  So what's wrong?  Rubbish copies?

So any recommendations for a good board with either very good wifi or ability to plug in an antenna?  I've even tried a few ESP32 boards from Amazon and they've all gone back for refund.

Aircoookie, 2020-05-26:
That exact board you bought works very well for me... Sorry that my top recommendation doesn't seem to work correctly for you!
Did you get it with the external antenna included? Not just any will work, it needs to be made for 2.4GHz. But if that was faulty, it shouldn't work directly after reset either... Did you try the binary or self-compiled?

fishbone-git, 2020-05-26:
I believe Wemos discontinued the version you linked, as it's now this
https://docs.wemos.cc/en/latest/d1/d1_mini_pro.html

I have one in the old form too, but no idea how to be sure it's legit. D1 minis (non pro) are top for me, though for wled I used wrover d32 minis (also hard to get 100% legit ones) due to the extra computing power. I get resets using wled and d1 minis, usually when interacting with the ui a lot at one time

Legsmaniac, 2020-05-26:
@Aircoookie - no, I didn't get the antenna's with the boards, I didn't get that far to needing them.  I was using the internal ceramic antenna for now.  But yes, I do know about them being designed for 2.4Ghz, no worries.  Oh, and I tried both binary and compiled.  I'd assume you'd expect the binary to work.

@fishbone-git - hmmmm, are the versions I linked discontinued versions, or are they just poor copies.  It's strange that I have no problems with the 12-F's, even running 2040 leds on the Matrix!  I'm gonna try attaching my own antenna to a few of them. There's several posts and TY Vids out there of people doing this with success.  I can but try one......

Legsmaniac, 2020-05-27:
> it needs to be made for 2.4GHz

Further to this..........
I watched a few videos showing people who just added a piece of wire to the end of the track on a NodeMCU without cutting any tracks, and demonstrated how effective it actually was.  I decided to try this since I had nothing to lose.  I ran a small program that gave RSSI readings in a terminal window and noted the readings before soldering on a piece of wire and again afterwards.  The result was impressive!  With no "aerial" I picked up only two SSIDs but with the "aerial" added, it was showing between 5 and 7 SSIDs, including the original 2 at between 12 and 32% increase in signal.

OK, then I decided to cut the tracks as shown in many of the tutes online and soldered on a coaxial antenna SMA holder for a 10cm 2.4Ghz 5dBi antenna and again noted the readings. I was surprised to note two things - 1) it wasn't as impressive as the piece of wire and 2) it received more SSIDs and better signal strengths with just the empty SMA connector then when I screwed on the antenna!

Just goes to show, once again as is quite often, *rules* are meant to be broken.  😉

raoulteeuwen, 2020-05-27:
> > it needs to be made for 2.4GHz
> 
> Further to this..........
> I watched a few videos showing people who just added a piece of wire to the end of the track on a NodeMCU without cutting any tracks, and demonstrated how effective it actually was. 

Could you share one or a couple of those video's?

Legsmaniac, 2020-05-27:
Sure.  The best simple single wire antenna video I saw was this one - https://www.youtube.com/watch?v=Mx0uEBPu7_8
Interestingly, I noticed my best signal was with the wire length 10cm.

And an example of adding a proper antenna, one of many like it, is this Instructables offering - https://www.instructables.com/id/External-Antenna-for-ESP8266/
There was loads like this, all very similar, on YouTube too.  Just search and ye shall find.

closed 2020-10-03 by stale[bot]