wled/WLED #949: Preset Cycle Resets?

Legsmaniac, 2020-05-26:
Is that correct?
If I set up a Preset Cycle to (say) cycle through 10 presets X minutes apart etc. etc. and then click Sync prior to ticking Preset cycle OR if WLED turns off OR if there's a power cut OR if the board resets for whatever reason, the settings revert to Cycle 1 to 5 and 1.2 seconds?  Is there a way to get it to stick or is this planned for 0.11.0/0.12.0?

There's only 212 more sleeps to Christmas 🤣 and I was planning on setting one ESP with (say) 10 presets of my liking to Preset Cycle through them all and Sync to the other ESP's in the house.  I was also hoping that it could be set so that when the ESP turns on (either via the App. Home Assistant, Alexa or whatever) it immediately starts with the first preset and cycles through as per Preset Cycle AND Syncs to the others which would also turn them on.  Would this be possible please?

Oh and not even the Sync stays selected, it always reverts to OFF next time the ESP is powered on.  😞

Aircoookie, 2020-05-27:
Yes, by default it is that way. The preset cycle feature is more of a "compromise" and I hope to replace it with a proper "playlist" feature soon where you can select the presets and durations for each one!
You can make WLED remember the values though. First of all, when you edit the first preset, last preset or time per preset, you will have to tick the preset cycle active once (you can immediately disable it again). Any other UI command will re-download the current state and override your values. After that, you can go into LED settings and tick `Set current preset cycle setting as boot default`. Then it will boot into your Preset cycle if it was on when saving the settings. If it was off, it will still remember first, last and time per preset.
For Sync, there is also a setting, just tick `Send notifications on direct change` in Sync settings :)

karen8000, 2020-07-10:
> Да, по умолчанию это так. Функция предустановленного цикла - это скорее «компромисс», и я надеюсь вскоре заменить его на соответствующую функцию «списка воспроизведения», где вы сможете выбрать пресеты и длительности для каждого из них!
> Вы можете заставить WLED запомнить значения, хотя. Прежде всего, когда вы редактируете первый пресет, последний пресет или время для каждого пресета, вам нужно будет один раз отметить активным цикл пресетов (вы можете сразу же отключить его снова). Любая другая команда пользовательского интерфейса перезагрузит текущее состояние и переопределит ваши значения. После этого вы можете зайти в настройки светодиодов и поставить галочку `Set current preset cycle setting as boot default`. Затем он загрузится в ваш цикл Preset, если он был включен при сохранении настроек. Если он был выключен, он все равно будет помнить первый, последний и время для каждого пресета.
> Для синхронизации есть также настройка, просто отметьте `Send notifications on direct change`в настройках синхронизации :)

Good afternoon friend, I want to know the habits of presets. I'll keep the presets and choose the interval and turn on the presets if the esp reboots then the busting settings are reset by default and the preset 📴 turned off, Even when the presets bust is active, the on-off button doesn't work 😔 until I turn off the presets
The firmware full poured on esp8266 01s flash drive changed to 4mb. 10 firmware version please 🙃 make another mode that if I turned on the scroll of the presets that esp memorized my choice 
Thank you 👍 😉 so much to you.

closed 2020-11-23 by stale[bot]