wled/WLED #950: LED PIN 3 Restricted?

Legsmaniac, 2020-05-27:
I just changed a source sketch to use LEDPIN 3 instead of pin 2, recompiled and uploaded.

Then if I try to set LED Count to (say) 1000 in the UI LED Settings page, a message informs me "!  Value must be less than or equal to 500" so I had no choice but to select pin 5 and use BitBang if I wanted to avoid using pin 2.

What can be changed to get around this?  Tried a sketch search for "500" and "value must be less" and it doesn't find anything.  🤔

Aircoookie, 2020-05-27:
Yep, LED pin 3 is unfortunately restricted to 500 LEDs. The problem is the DMA hardware driving method used on (just) that pin. It works well, but uses **4x (!)** as much RAM memory as the UART hardware driving on pin 2 and the bitbang driver on all other pins.

Legsmaniac, 2020-05-27:
FOUR times the memory?!?!?  Jeebus!!  I never knew that!!  Yet I managed to set 2040 leds using pin 3 earlier today and it was running fine so far.  Oh well, bitbang it is then.

Thank you so much for all your answers to our questions, I've learnt so much in such a sort time, stuff I honestly never realised.  Honestly, you are one of the few devs who take the time to answer our (stupid) questions in a really friendly manner.  It's so refreshing to see.  Keep up the good work.  😉  And many thanks again.

Aircoookie, 2020-05-31:
Added to FAQ 🙂

closed 2020-05-31 by Aircoookie