wled/WLED #951: Is the traffic light effect broken?

spiff72, 2020-05-27:
I just noticed that the traffic light effect seems to behave in a non-intuitive way:
My expectation is that it would operate like a traffic light:
Red->Green->(short yellow)->Red (and repeat)

What it actually does is:
Red->(short Red/Yellow at the same time)->Green->(short yellow)->Red (and repeat)

Is this the way traffic lights behave outside the US, by chance, and my assumption about how traffic lights operate is based solely on my own environment?  ;-)

raoulteeuwen, 2020-05-27:
Like this: https://www.youtube.com/watch?v=xqOcw6XeOeg ?

Legsmaniac, 2020-05-27:
Yep, same as in the UK  🙂
The way Traffic Lights SHOULD be.  😝

I guess you'll have to submit a request for US Traffic Lights  🤣

spiff72, 2020-05-27:
Someone should fix that traffic light, since it is CLEARLY broken! 

Seriously - I had no idea that they behaved differently outside of the US.  I kind of figured since no one had pointed out what seemed like an obvious error, that there must be some reason for the red/yellow combo.

spiff72, 2020-05-27:
I will add, though, that putting the yellow in there before green DOES make some sense - sort of a warning for those that just got the green light to proceed with caution in case some idiot is about to run the red.

Aircoookie, 2020-05-27:
As a German, naive me actually though the whole world would have this sequence 😂 
Will add an option to the effect soon for a sequence without the red/amber "prepare to go" signal.