wled/WLED #955: Connecting WLED NodeMCU to Home Network

nonisclan, 2020-05-29:
I am a complete beginner to this, I followed all the videos I can find, but when I try to connect the WLED-AP to my Home Wifi it does not appear that they are connecting.

This may be a really dumb question but on my laptop, should I be able to connect to the NodeMCU and be able to browse the internet at the same time?

huggy-d1, 2020-05-29:
Your question.. If you connect via wifi to WLED (on your NodeMCU for example), and have no other wired or wireless connections on your laptop PC, then no, the internet should not be accessible.

On a smartphone though, you should be able to connect to both if you have data coverage via cell.

raoulteeuwen, 2020-05-29:
@nonisclan i hope @huggy-d1 's answer does cover your situation. If not:

- if your WLED controller has no info or way of connecting to your wifi, it starts acting as an access point itself. In this last mode, you can connect to the device, but you won't be able to use internet
- if your WLED is able to connect to your home wifi, but you are still trying to access the WLED controller in it's access point mode, you won't be able to connect
- if your WLED is able to connect to your home wifi, you should be able to see it on your router as one of the devices connected to it, including its IP address. You can also install the app Fing on your phone or tablet and scan your network to find out what devices are connected to it. If your device is connected to your home Wifi and WLED is, you should be able to connect to WLED, for instance using an IP address in the range of your home Wifi (192.1.168.x maybe?).
- you could in all situations try the WLED app on your mobile device and see whether you can connect to your WLED via the app

nonisclan, 2020-05-29:
Thank you both so much. I figured out the problem.

closed 2020-05-29 by nonisclan