wled/WLED #956: 433 MHz receiver support?

vtni, 2020-05-29:
Hi there,
it would be great if we could turn WLED on/off or change effects through a 433 MHz remote. Is something like this planned?

Aircoookie, 2020-05-29:
Hi,
no that isn't planned right now for the core of WLED. However you can add support using an usermod :)

Legsmaniac, 2020-05-29:
Just curious but why would you need that when you have the app?  🤔

vtni, 2020-05-29:
> 
> 
> Just curious but why would you need that when you have the app? 🤔

I wanted to build a wireless button (without any hub) that isn't connected to the wifi.

Legsmaniac, 2020-05-29:
Fair point.  
Just wondered because you'd still need to be within range for 433Mhz to work and so the same could be said for using AP which wouldn't rely on WiFi connection to your router.  Kinda the same thing in a way but with all the functions.

vtni, 2020-05-29:
That's right and my first idea was to use an ESP as a button, however if it is not connected all the time, it will take some seconds until connected to the wifi. And i wanted to avoid delay. Deep Sleep would also not help in that case.

Legsmaniac, 2020-05-29:
> it will take some seconds until connected

Ah yeah, good point.  Fair enough.  Thanks.

closed 2020-10-04 by stale[bot]