wled/WLED #958: Color Picker issues

OMGchad, 2020-05-31:
First off, I really love WLED. Huge thanks for making and maintaining this. 

I am having an issue closing the exact color I would like with the current color picker and would love to see if there is a workaround. I am going for a hot pink (R:255, G:0, B:255) and cyan (R:0, G:255, B:255) look. With the current color picker, I can assume I am close to these values but it would set my mind at ease knowing they were directly dialed in. Is there a way to set these values?

It would be a great feature if the color picker had a field for either HEX color or RGB values. Another feature would be to change the options of easy access colors under the wheel. One frustration of mine is the color that looks like cyan does not have enough blue in it. 

[image]

I don't think it matters but I am running WLED 0.10.0 (build 2005030) on a ESP8266

OMGchad, 2020-05-31:
I have figured out a workaround, sorry for the unnecessary issue. 

Workaround
----
Useing [HTTP request API](https://github.com/Aircoookie/WLED/wiki/HTTP-request-API) I was able to just send `http://[wled-ip]/win&R=255&G=0&B=255` and `http://[wled-ip]/win&R2=255&G2=0&B2=255` to the device. That way I had the colors I wanted. Then I used a [preset](https://github.com/Aircoookie/WLED/wiki/Presets) to save to color options.

closed 2020-05-31 by OMGchad

Aircoookie, 2020-05-31:
No worries, I plan on adding an HEX color input field to the UI soon :) Tracking in #506