wled/WLED #960: Request: special on-dim-off mode

mike811, 2020-06-02:
Hello,

many thanks for this great projekt.
I'm kind of lost in the many features, macros, presets, etc. so sorry if this is already implemented...

I need to dim my light OFF in a special way: 
- It should first turn all LEDs (of a segment) RED in lets say 2 seconds and then
- reduce brightness to minimum and then
- switch all LEDs off sequentially from one end in 2 seconds (lets say we have 100 LEDs: 100 LEDs low red brightness, 99 LEDs, 98 LEDs, etc.)

Turning ON would be the reversed sequence from above.

I know, that I could do this by an external interface but it would be nice having it build in.

Aircoookie, 2020-06-02:
Hi, awesome that you like WLED!
This is a very specific way of turning off the LEDs which wouldn't really make sense to implement as a general feature. What I would suggest is an usermod (which requires a little bit of coding) that times this sequence for you. For turning off all LEDs in sequence, the Wipe effect is ideal, like it is done in the [Stairway usermod](https://github.com/Aircoookie/WLED/blob/master/usermods/stairway_wipe_basic/wled06_usermod.ino)

Alternatively, I have planned a playlist feature where you'll be able to create this sort of sequence. I don't yet know when I'll get around to implementing it though 😅

huggy-d1, 2020-06-02:
@Aircoookie do you have anything in the works, a sketched up architecture or other-wise, for the playlist feature?  Like effect[0] & effect[0].{settings}, delay (n[0] msec), ..., effect[n] & effect[n].{settings}, stored as a preset for example?

Aircoookie, 2020-06-02:
Yep, I got the architecture and started some implementation as a test a few months ago, but it is far from finished. In the simplest form, you have a JSON list of preset to cycle through and the duration for which to stay on each preset. But for it to work I need to finish the filesystem first :)

closed 2020-10-09 by stale[bot]