wled/WLED #961: getting flash read error

kwynns, 2020-06-02:
Serial output loops,

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

I've tried flashing it to different areas and it doesn't seem to do the trick.

I've tried NodeMCU-PyFlasher-4.0-x86 and then just started using esptool

This seems to suggest that it could have to do with wifi/OTA? 

I tried, esptool.py --port COM5 --baud 115200 --after no_reset write_flash --flash_mode qio 0x1000 WLED_0.10.0_ESP32.bin --erase-all

as well as dio 

[image]

kwynns, 2020-06-02:
just saw this, https://github.com/Aircoookie/WLED/issues/441 trying that now

kwynns, 2020-06-02:
hmmm, ran

```
PS C:\Users\kelse\Documents\wled> esptool.py --port COM5 write_flash 0x10000 C:\Users\kelse\Documents\wled\ESP32.bin --erase-all
esptool.py v2.8
Serial port COM5
Connecting........____
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a4:cf:12:25:d7:24
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.1s
Compressed 1076672 bytes to 618512...
Wrote 1076672 bytes (618512 compressed) at 0x00010000 in 55.3 seconds (effective 155.9 kbit/s)...
Hash of data verified.
```

and still getting that loop...I'm baffled.

Aircoookie, 2020-06-02:
Hi! Sorry for the trouble! Try flasing the `esp32_bootloader.bin` from the 0.9.1 release assets and then flash again :)

kwynns, 2020-06-02:
haha, I just came here to say that's exactly what I tried, and it worked!

Thanks...at the least I wanted to get this issue up so it could be searchable for others. If you'd like me to test anything else just let me know. I had bought a dozen cheap wroom esp32s off aliexpress awhile back. I also have a wroover and had the same issue (I was able to flash from arduino ide on the wroom previously)

closed 2020-06-02 by kwynns

Aircoookie, 2020-06-03:
Awesome that solved it for you :) I've also added the bootloader to the 0.10.0 assets now.
With Arduino IDE the bootloader is flashed automatically, so this issue only occurs when flashing the binary to a brand new ESP32 for the first time without having uploaded an Arduino sketch before.

obol2, 2020-06-22:
Hello, I have the same problem as above. I did the following steps:

1. Flashed bootloader with nodemcupyflasher  (erase flash, 115200, dio)
2. Disconnect esp32 and reconnect
3. Flashed wled.bin for esp32 (NO erase flash, 115200, dio)
4. disconnected and reconnect, in arduino ide serial monitor I get the same error:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

dc-smartindustry, 2020-11-17:
Exactly the same here. switched to 0.9.1 but got the same error. On all devices. No idea where to look further.

Claustn, 2020-11-23:
@dc-smartindustry I just ran into the same issue, when noticed that it tried to write both things with 0x0000 it failed, this made it work here

esptool.py --port COM3 --baud 115200 --after no_reset write_flash --flash_mode dio 0x00000 C:\wled\esp32_bootloader.bin --erase-all

Reboot device

Run
esptool.py --port COM3 --baud 115200 --after no_reset write_flash --flash_mode dio 0x10000 C:\wled\WLED_0.10.2_ESP32.bin

Notice the 0x0000 and 0x10000