wled/WLED #963: WS2811 usage

Partizan7676, 2020-06-03:
WLED is works fine with my NODEMCU board and 2812 stripe, but 2811 doesnt. What i need to do for using 2811 strip? If i connect 2811 strip to power supply and controller it light up with white color whole stripe(no matter how many leds defined in config just all), it is same when i connect it to power without controller.

Aircoookie, 2020-06-03:
Hi! The 3.3V signal of the ESP8266 is not enough to drive many WS2811, which will just show pure white. You have 3 options to fix this :)
1. Use a WS2812  as first LED and hook up the WS2811 to its output
2. Try this [diode trick](https://hackaday.com/2017/01/20/cheating-at-5v-ws2812-control-to-use-a-3-3v-data-line/)
3. Most reliable: Use a 3.3V->5V level shifter (I recommend the SN74AHCT125N)

Partizan7676, 2020-06-03:
2812 is 5V, but 2811 is 12V. I am using DCDC inverter to get 5V for controller and it works fine with 2812 stripe. If i replace 2812 stripe with 2811 get pure white. When i connect 2811 stripe just to power supply, without any controller, i get pure white colour, if do that with 2812 it stays off. Is it strange or not?

sansillusion, 2020-06-04:
What Aircoookie is explaining is, the esp is sending signals at 3.3V on the
DATA pin, the ws2812 being 5V work because the signal is strong enough
(over 50% of 5v) but for 2811 they expect a signal of 12V but would
probably work at 5V but often 3.3V is not enough to get a clear signal
since it's way under it's 12V. Home I made it a little more clear.

nelzon101, 2020-06-05:
Hello

i use Wemos D1 mini with WLED and a 12v WS2811 without any changes and problems! 
The ground from Wemos and WS2811 must be connected **together**!

Partizan7676, 2020-06-05:
@Aircoookie THX a lot! Problem is solved! As described in driver datasheet level must be 3.5-5.5V. so i made trick with diode. and it works just fine! if you need scheme i can send it to you via email for wiki extend.

Aircoookie, 2020-06-05:
Awesome to hear the diode trick worked :)

closed 2020-06-05 by Aircoookie

sbali620, 2022-02-13:
> @Aircoookie THX a lot! Problem is solved! As described in driver datasheet level must be 3.5-5.5V. so i made trick with diode. and it works just fine! if you need scheme i can send it to you via email for wiki extend.

Yes please tell me I want the same? How can I use it with my Wled ws2811?