wled/WLED #966: ESP32-S2 support ?

johnall203, 2020-06-05:
Hello noob here.  I was wondering if the newer ESP32-S2 board was supported ? Has anyone tried it and is the installation process the same as the regular ESP32 ?  I figured if I was going to try it but wanted to be sure before making a purchase.   Thank you for the help ! John

Aircoookie, 2020-06-06:
Hi!
Seems like the ESP32 Arduino core team is working on it, but the ESP32-S2 doesn't appear to be 100% compatible with the existing sources and binaries: https://github.com/espressif/arduino-esp32/issues/4066

So most likely there is no support for the ESP32-S2 yet until it is supported in the Arduino Core.
Will use this issue to track ESP32-S2 support specifically in regards to WLED :)

vetezeTE, 2020-10-15:
Any updates on the ESP32-S2 boards? I bought 4 of these not really thinking it would be an issue but I haven't been able to get WLED to work with them.

vetezeTE, 2020-10-15:
Unexpected error: ESP Chip Auto-Detection failed: Unexpected UART datecode value 0x00000500.

misery, 2021-01-01:
I found/added/ experimental support for my ESP32-S2 in ESPHome: https://github.com/esphome/esphome/pull/1433

As WLED also uses platformio it could be adopted easily until it is full supported?

```
[env:esp32s2]
board = nodemcu-32s
board_build.mcu = esp32s2
board_build.partitions = partitions.csv
framework = arduino
platform = espressif32@1.12.4
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.2
```

misteraddons, 2021-10-05:
I would also like to know if there's any update on this? ESP32-S2 has been out for a year now and it seems like the future. Thanks :)

blazoncek, 2021-10-05:
Discord.

lsiudut, 2021-12-27:
FWIW I managed to build WLED for ESP32-S2 with minor (yet dirty) patches over LittleFS (which is part of `arduino-esp32`) and ESPAsyncWebServer (mbedtls was apparently bumped to newer version). Haven't had much time for detailed testing but it seems to be working just fine (w/ ws2811 at least).

Given that I care about newborn baby now I don't expect myself to have time to familiarize myself with the codebase to submit a high quality PR. I don't know how far has the project went with the issue, maybe there's cleaner way of making it buildable already, but in case there isn't, here are gists with patches that make it work:

- for WLED: https://gist.github.com/lsiudut/af6fcd9c9c2e9d30e777f61a77604a7e
- for ESPAsyncWebServer: https://gist.github.com/lsiudut/1722a69cacef166539ac4c88d117c25f

misery, 2021-12-27:
Funny... nearly at the same time: https://github.com/Aircoookie/WLED/pull/2452 😉🙃

lsiudut, 2021-12-27:
Ah, there you go. I haven't even check PRs. Disregard then :).

KishCom, 2022-05-09:
Thought I'd chime in here with some recent experience. On my ESP32-S2-saola dev boards:

Using it to drive 256 Dotstars (APA102) seems to work just perfectly! (`-DUSE_APA102`)

Using it to drive 600 WS28XX LEDs, I start [hitting the RMT error described here](https://github.com/Aircoookie/WLED/pull/2452#issuecomment-1005982573).

kudlav, 2022-07-07:
Tried to build 0.13.1 with `esp32s2_saola` target. Switching the presets doesn't work. ESP closes the WebSocket connection (Connection Closed: 1005 ) after sending the request to switch a preset. Preset is not switched, and a new WebSocket connection is immediately created. JSON and HTTP interfaces doesn't change preset either, without any error codes or messages.

blazoncek, 2022-09-15:
Anyone interested may try https://github.com/blazoncek/WLED/tree/esp32-s2 
Please use this as PIO environment:
```ini
[env:lolin_s2_mini]
platform = espressif32@5.1.1
board = lolin_s2_mini
board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv
board_build.flash_mode = qio
upload_speed = 460800
upload_port = /dev/cu.usbmodem01
monitor_port = /dev/cu.usbmodem01
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags} ${esp32s2.build_flags}
  -DBOARD_HAS_PSRAM
  -D ARDUINO_USB_CDC_ON_BOOT
  -D WLED_USE_PSRAM
  -D WLED_WATCHDOG_TIMEOUT=0
  -D CONFIG_ASYNC_TCP_USE_WDT=0
  -D LEDPIN=16
  -D BTNPIN=0
  -D HW_PIN_SCL=35
  -D HW_PIN_SDA=33
  -D HW_PIN_CLOCKSPI=7
  -D HW_PIN_DATASPI=11
  -D HW_PIN_MISOSPI=9
  -D STATUSLED=15
  -D WLED_DEBUG
  -UWLED_USE_MY_CONFIG
lib_deps = ${esp32s2.lib_deps}
```

michael-blue2, 2022-10-05:
@blazoncek Thank for your efford for the work that you did! I build the bin file using visual studio with the settigns you suggested in the PIO. 
I just need some help with flashing. Even when running esphome, i was only able to make them work with this method:
while holding the O button connecting to pc. then press once the rst button and after some sec release the O. that would take the esp s2 into download mode. after I use esptool to erase the flash: 
   esptool ip com9 erase_flash
without doing anything on the esp, I would then go to esphome and flash it. It always needed 2 times. first time pressing the flash gives an error. second it loads and everything from there its all good (update online and everything)
When I was trying to make it work with esphome i did try to flash bootleader first and then my bin file but no luck. 
So now back to wled. 
I do have the bin file ready and i have tried flashing with esptool( it does flash succesfully but wifi ap doesnt work. any other ways that i have tried doesnt work. Any ideas?

blazoncek, 2022-10-05:
What worked for me (until I understood how things work) I flashed MiroPython 1st using the instructions on their page using `esptool`.
Once MicroPython was up & running. I flashed `firmware.bin` from PIO build folder at the address 0x10000.
You may skip the MicroPython if you flash with PIO but make sure you have correct partition map .CSV file.

blazoncek, 2022-10-05:
Unfortunately running WLED on S2 is still very much hit & miss. If you do not access web UI it will be running ok, but frequent UI access will cause it to loose wifi connection or reboot.
It may be a flaw in WiFi or TCP stack or AsyncWebServer incompatibility.

softhack007, 2022-10-21:
Just stumbled over this bug report for CDC on S2: "Issue realted ESP32-S2 CDC to the Serial.setDebugOutput(true)"
https://github.com/espressif/arduino-esp32/issues/6766#issuecomment-1251546266
seems that debug out from parallel tasks will cause a lock-up.

Germaine-Sail, 2022-11-07:
> Anyone interested may try https://github.com/blazoncek/WLED/tree/esp32-s2 Please use this as PIO environment:

Apologies if I'm asking a silly question but the above link seems not to work, I'd like to compile this code for an ESP32-S2, I take it the above link use to navigate to a WLED clone that was tweaked for the above IC.

Thanks
Germaine

blazoncek, 2022-11-07:
It has been since merged into main WLED.

michael-blue2, 2022-11-07:
> It has been since merged into main WLED.

How do you guys install wled on the s2? I had lots of trouble just installing the esphome as well. but managed to find a way using their online installation but just needed 2 times of the sequence

blazoncek, 2022-11-07:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
As a side note: it is a waste of time ATM to try anything on S2.

tanner0101, 2023-01-25:
I was able to get WLED running on [my S2 mini board](https://www.amazon.com/dp/B0B28LMBKD?ref=ppx_yo2ov_dt_b_product_details&th=1) using the `lolin_s2_mini` env. 

To do this I just followed the [normal install instructions for compiling binary](https://kno.wled.ge/advanced/compiling-wled/) but overriding default envs to the s2 one @blazoncek added.

```diff
; default_envs = esp32s2_saola
+default_envs = lolin_s2_mini
```

I was a little confused on whether I needed to flash a bootloader or not. I ended up flashing the ESP32 v4 bootloader as the docs recommend. I did this using `esptool.py`.

I also tried flashing the binary with `esptool.py` but ultimately just used the "upload" button in PlatformIO as that was easiest. Had to reset things a few time on my bread board but then it worked! I'm able to connect over WiFi and everything.

[image]

Note that to get the ESP32 to connect over USB I had to hold the two buttons while for 1-2s while plugging it in.

I've read there may be issues but I think it might be worth allowing this via the installer website maybe behind a warning just so its a bit easier to try for folks.

Hoping to learn more about how WLED and PlatformIO work soon so that I can contribute :) Awesome work on this project I love it.

blazoncek, 2023-01-25:
ESP32 variants S2, S3 and C3 are partially supported but still very much experimental.
Partially due to buggy Espressif core libraries and lacking support in underlying Arduino platform libraries.

closed 2023-01-25 by blazoncek

softhack007, 2023-01-25:
@tanner0101 thanks for sharing the details on how you made your board work.
I agree that currently the best approach is to build yourself in platformio. Pre-build .bin files are often not working on other boards.

> I was a little confused on whether I needed to flash a bootloader or not. I ended up flashing the ESP32 v4 bootloader as the docs recommend. I did this using esptool.py.

The standard " ESP32 v4 bootloader" __cannot work__ on -S2, as the CPU instruction set is different. You were lucky that platformIO upload also replaces invalid bootloaders.