wled/WLED #968: Madmapper x WLED - E1.31 problem

pk88888, 2020-06-06:
Thank you for this project, great solution.

I have a problem transferring data to the device. If I run the E1.31 option, after restart the device will start receiving data once and once not. The moment I run Artnet there is no problem and everything works fine. If I switch to E1.31 again, the device stops receiving data again and goes into static lighting mode. Has anyone had a similar problem? I have the latest version of the software installed.

Aircoookie, 2020-06-06:
Hi, awesome that you like WLED!

You will need to reboot each time after you switch between E1.31/Art-Net but it sounds like you've done that. How many LEDs are you using? I am surprised about this as there is very little difference between the Art-Net and E1.31 implementations. What do you mean by it receives data `once and once not`? Does it start receiving correctly but then stops completely? Does it stop for a few seconds and then resumes receiving? Hope we can get this to work for you :)

pk88888, 2020-06-17:
Hi!

I apologize for the delay in answering the question. The problem is that if I choose ArtNet in WLED and send such data from madmapper then after restart, the device will go into the data receiving mode and everything works correctly. At the moment when I change  WLED to sacn mode, then after changing the sent data in madmaper to sacn and restarting the WLED device, nothing happens. After a few attempts to restart everything starts working properly. The number of pixels does not matter.

Aircoookie, 2020-06-17:
Interesting that that happens! I have absolutely no idea whats causing it though, the WLED udp interfaces should be completely re-initialized after every reboot and should work with the selected mode. I only have two theories:
- madmapper sends out sacn data to the Art-Net port (would be weird)
- the universes don't match (Art-Net default start is 0, in E1.31 sacn it is 1). But in that case it shouldn't start working after a bit more restarting.

So I'm not sure what exactly is going on here but luckily this won't be a major issue as usually you choose either Art-Net or E1.31 and don't swap between them often.

pk88888, 2020-06-18:
I will look for a solution and let you know as soon as we solve the problem.

I need to test other software. I've tested different computers and behaved the same on all of them. I have also tested it on several ESP devices and they all behave the same.

closed 2020-10-24 by stale[bot]