wled/WLED #971: Anyone succeeded in building WLED for LPD8806?

user-51, 2020-06-07:
I set a custom config in platformio_override.ini:

```
[env:d1_mini_custom]
board = d1_mini
platform = ${common.platform_latest}
board_build.ldscript = ${common.ldscript_4m1m}
build_flags = ${common.build_flags_esp8266} -D USE_LPD8806
```

Building fails with the errors
```
...
.pio/libdeps/d1_mini_custom/NeoPixelBus_ID547/src/NeoPixelBus.h:361:10: error: no type named 'SettingsObject' in 'class Lpd8806GrbFeature'
...
.pio/libdeps/d1_mini_custom/NeoPixelBus_ID547/src/NeoPixelBus.h:133:106: error: 'SettingsSize' is not a member of 'Lpd8806GrbFeature'
...
.pio/libdeps/d1_mini_custom/NeoPixelBus_ID547/src/NeoPixelBus.h:389:57: error: 'pixels' is not a member of 'Lpd8806GrbFeature'
...
```

Aircoookie, 2020-06-10:
Confirmed that LPD8806 builds seem to be broken. Will look into it!

User2814, 2020-09-20:
Was this bug fixed? Did anyone get LPD8806 Strip working (compiling)? If yes, what version of WLED???
WLED version 0.10.2 is not working with LPD8806

Aircoookie, 2020-09-20:
@User2814 I was able to reproduce this issue with a simple example sketch of the NeoPixelBus library responsible for driving the LEDs. I've opened an issue https://github.com/Makuna/NeoPixelBus/issues/368 in the library repo, hopefully it is resolved soon :)

Aircoookie, 2020-09-20:
Makuna was lightning fast with fixing the issue! There also was a small issue in WLED with the `PIXELFEATURE4` not being defined.
So make sure to grab both the latest master of NeoPixelBus (it's automatic if you use platformio) and the latest WLED master and it should work again!
Since I don't have LPD8806's on hand for testing it would be cool to have feedback whether they work as expected :)

User2814, 2020-09-21:
> Since I don't have LPD8806's on hand for testing it would be cool to have feedback whether they work as expected :)
Thank you @Aircoookie !!! Thank you, now its working. What I saw at first look is that not all Patterns are working, but this can be also my disability to work with the app. The basics are working without Problems. Thanks for your great work!!!!

Aircoookie, 2020-09-21:
@user-51 awesome, thanks for the feedback! If some effects don't appear to work, try moving the Speed and Intensity sliders, and they should look better. Or try selecting a palette or different colors. If there are still non-working effects, you can write me their names and I'll look into it.

closed 2020-09-21 by Aircoookie

User2814, 2020-09-22:
@Aircoookie thanks, working great!!! Problem was that I didn't select the right "123" colors. Didn't expect its working so well 👍👍👍👍👍 Thanks for great work!!! 
Is there any chance to put there my own effect pattern (I have already my own patterns (.ino file) fitting to the Timeline of a music song) ?
Or where are the patterns stored?

Aircoookie, 2020-09-22:
@User2814 all effect functions are in the file `FX.cpp`. Of course it is possible to add custom animations, but keep in mind that they require some slight modifications (for example, to set a pixel it is `setPixelColor(uint16_t index, uint32_t color)`, which is different from native FastLED or NeoPixelBus-using code). You can take a look at the other effects to get an idea of how it works :)