wled/WLED #974: For potential future inclusion - playback recordings

Legsmaniac, 2020-06-08:
First it was TPM2.NET then it was Glediator/Serial/TPM2 requested.
However, both require direct connection/transmission via a PC so I decided to try using an SD Card Reader connected to my ESP8266 to play Jinx/Glediator Matrix recordings.  I have to say, it worked very well.  I connected` MISO to GPIO12`, `MOSI to GPIO13`, `SCLK to GPIO14` and `CS to GPIO4` and ran this simple stand-alone sketch....

`#define FASTLED_ESP8266_RAW_PIN_ORDER
#include <FastLED.h>
#include <SPI.h>
#include <SD.h>

#define NUM_LEDS 2040
#define DATA_PIN 2
#define CD_PIN 4
#define maxPower 10000
#define maxBrightness 32
String fileName = "Matrix.dat";
unsigned char matrixWidth = 60;
unsigned char matrixHeight = 34;

File jinxData;
CRGB leds[NUM_LEDS];

void setup()
{
  FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);
  FastLED.setMaxPowerInMilliWatts(maxPower);
  FastLED.setBrightness(maxBrightness);
  fill_solid(leds, NUM_LEDS, CRGB::Black);
  FastLED.show();

  pinMode(CD_PIN, OUTPUT);
  digitalWrite(CD_PIN, HIGH);

  SD.begin(CD_PIN);
}

void loop()
{
  jinxData = SD.open(fileName);

  while (jinxData.available())
  {
    jinxData.readBytes((char*)leds, NUM_LEDS * 3);
    CRGB tmp[matrixWidth];
    for (int i = 1; i < matrixHeight; i = i + 2)
    {
      for (int j = matrixWidth; j > 0; j--)
        tmp[(matrixWidth - j)] = leds[i * matrixWidth + j - 1];
      for (int j = 0; j < matrixWidth; j++)
        leds[i * matrixWidth + j] = tmp[j];
    }
    FastLED.show();
  }
  jinxData.close();
}`

It played the file Matrix.dat I had set up as a recording on Jinx! to my 2040 led matrix so I'm thinking this is undoubtedly a better way for me than TPM/Serial.  I realise you're including Playlists in a future edition of WLED but I thought what a great way this could be used for *quick change playlists* by simply swapping the SD Card.  Just a thought.  Obviously the main idea would be for the forthcoming Matrix inclusion?

On another note, after all this, I'm now no longer sure I want to use my 2040 led Matrix in my Christmas Project 2020.  😂🙄🥴😕

harueg, 2020-06-12:
Thanks , will try it. This is stand alone, w/o WLED ?

Legsmaniac, 2020-06-12:
> This is stand alone, w/o WLED ?

The above code is stand-alone, yes.
I'm hoping @Aircoookie will consider it or something similar for future inclusion into WLED, to add SD Card functionality to the excellent WLED for recording playback.   🙃

Aircoookie, 2020-06-13:
Thank you for the standalone sketch!
Integrating this into WLED should be quite easy and I will do so right after filesystem support for presets. I believe SD cards could be a good way to store the preset file as well optionally as it will be easier to recover if the WLED installation somehow fails.
The only thing I wonder about is why this sketch doesn't have any sort of timing - it seems like it plays the data as fast as possible.
Let's track this in #627 :)

closed 2020-06-13 by Aircoookie

Legsmaniac, 2020-06-13:
> The only thing I wonder about is why this sketch doesn't have any sort of timing - it seems like it plays the data as fast as possible.

You are absolutely correct.  The sketch was modified for myself and my Matrix of 2040 leds.  There *was* a delay in original sketches I modified this from.  It was quite simply `delay(20); // set the speed of the animation. 20 is appx ~ 500k bauds` placed after `FastLED.show();`  I found I didn't need it as recordings were playing back at the speed I recorded them in Jinx anyway.  I also added the code to fit my snakeline layout since Jinx! didn't allow for any settings when making recordings which was very strange.