wled/WLED #975: Request : Implement HEX in JSON-API

DutchmanNL, 2020-06-08:
For ioBroker integration I currently calculate RGB to HEX and vice versa as most color pickers and iOT systems are using HEX values.

Can you please integrate the HEX values also into the JSON-API, as far I see the set command in HTTP is already working but I would like to get also the data in HEX values by API instead of calculating it afterwords.

Aircoookie, 2020-06-09:
I should have probably considered using HEX for the color format as in `["FFA000","000000","000000"]` from the get go. The RGB Integer format `[[255,160,0],[0,0,0],[0,0,0]]` is easier to parse, but harder to read and it also has a variable length, so HEX would have been more consistent. It's also a matter of preference though and of course the conversion is not hard.

While I could add the API accepting HEX values as the input, outputting them is likely not practical at this point (for backwards compatibility, we can't just switch over since it would break compatibility with existing sofware) and I also don't want to add both, because that would be redundant information and we can't afford the JSON API response becoming longer (it would mean we have to reduce the number of max. segments). Right now I don't have a good solution, but I'll think some more if there is something that can be done.

Legsmaniac, 2020-06-10:
> as in `["FFA000","000000","000000"]`

Forgive me if I'm not understanding this correctly but why "as in" your example and not simply as in `["AA126A"]` (which equals `["R170,G18,B106"]` ?
I didn't think it was possible for 16 million shades of each colour with LED strips and even if there was, would they be that discernible?  @DutchmanNL s statement "I currently calculate RGB to HEX and vice versa" would suggest to me that he's obviously using the current system of 255 values (00 to FF) of each colour since that is what it is now.
Therefore, wouldn't it be easy to add a HEX entry field in the UI together with the RGB entry field?  It only requires a simple calculation to convert between the two as I'm sure you realise.  🙂

DutchmanNL, 2020-06-10:
Hmm ok clear to bad, in basic I can use the http api to send hex and convert RGB from JSON back to hex for iobroker.

Native would be better but also understand the need to keep the JSON light

Aircoookie, 2020-06-10:
@DutchmanNL well I can at least add the capability to WLED to accept HEX colors via JSON, so that you and others don't have to use the HTTP API just for that. Just outputting them is an issue as it bloats the state object up a bit.

@Legsmaniac you are correct, a single HEX color has the format `"FFA000"`. There is an array of 3 of them, because each segment currently supports a primary, secondary and tertiary color :)

Legsmaniac, 2020-06-10:
Ahhhhhh! Thank you for the clarification. I knew I must have been missing something! Doh!  

In that case, I'd say stick with the current RGB system and avoid bloat. There are plenty online RGB/HEX conversion calculators and colour pickers we can use. There's even apps for them!   🙃

Aircoookie, 2020-11-20:
WLED now accepts HEX colors in json api commands in latest master :)
Example: `{"seg":{"col":["FF0000","00FF00","0000FF"]}}`.
Will be added to docs tomorrow.

Still holding back on breaking changes (state object will continue to send out the RGB array format for now), but maybe that could be changed in 0.12. If I absolutely have to make breaking changes, I want to consolidate them so that clients don't have to adapt too often :)

DutchmanNL, 2021-01-04:
@Aircoookie awsome need to test it how it works out and integrate

fguchelaar, 2021-01-24:
I can't seem to get this to work for _individually_ addressing LEDs. Is that to be expected? 

For instance: 
```
{"seg" : { "i" : [ "FF0000" , "00FF00" , "0000FF" ] } }
```

Aircoookie, 2021-07-20:
@fguchelaar thank you for the suggestion! Hex format for individual LED control is now supported in version 0.13.0-b2 :)

Closing this issue, as no breaking changes to the data _sent_ by WLED are intended for the time being. I managed to resolve the memory issue internally without changing the API. Further discussion is however encouraged :)

closed 2021-07-20 by Aircoookie

DutchmanNL, 2021-08-06:
cool than I can remove that translation from my code :)

DutchmanNL, 2021-08-09:
looks good, please test