wled/WLED #978: Addresable RGB strip and CCT strip support

J-Wrobel, 2020-06-08:
Hello, 
I am thinking about making a lamp with one addressable RGB(W?) strip and one CCT strip. Does WLED support this kind of setup? It would mean driving two analog channels for warm and cold white of CCT and addressable RGB and maybe W. Those two 'modes' will not have to run at the same time. The idea is to have sort of daylight lamp and a mood lamp all in one.

Aircoookie, 2020-06-11:
Hi!
This is supported with some minor modifications:
- Recompile with `WLED_USE_ANALOG_LEDS` enabled in NpbWrapper.h
- Hook up the CCT strip inputs to GPIO 5 and 12
- Create two segments, the first is just the first LED and the second are all the rest
- With the color of the first segment you can control the CCT strip (red = warm white, green = cold white)
- With the second segment the RGB can be controlled normally.

It's a bit of a bodge, but should work! I plan to improve the LED system to allow such a config in a more convenient way at some point soon.

J-Wrobel, 2020-06-11:
Splendid! Thanks for the tip. I will keep an eye on updates though.

closed 2020-06-11 by J-Wrobel

Ramblurr, 2021-06-23:
@Aircoookie Have you been able to do this yet?

I would like to drive 4x analog CCT (warm white and cold white) strips. Is that possible?

Aircoookie, 2021-06-23:
@Ramblurr kind of. You can now define up to 8 analog CCT strips on ESP32, but they are still driven from the red and green color channels.

lanwin, 2022-02-12:
Hi, maybe that is not the right place for asking, but I am want to build a hanging light above my desk and with two CCT LED Strips (one down and one up) and control it with WLED. But I did not yet find any good resource how to connect them properly to an ESP32 and what to consider when doing it. The strip is one are 5m 24VDC SMD5050. I plan to use 2x1m on each direction and want to control both directions separately. 
Did you have any hint or resource I could look into?

J-Wrobel, 2022-02-12:
I would point here for nice resources on LED lights https://quinled.info/