wled/WLED #980: Enabling/disabling the checkbox "Receive UDP realtime" has no effect

alborzs, 2020-06-09:
I am trying to control a led strip from two interfaces, **E1.31 (DMX)** (via xlight) and **MQTT** using an ESP32.

There is a checkbox in **`Sync setup > Realtime > Receive UDP realtime`** but enabling/disabling that checkbox seems to have no effect. The animation from **E1.31 (DMX)** keeps playing regardless of the toggled checkbox.

That seems to be bug, but maybe I have missed something and that checkbox actually means something else, in that case please explain it to me.

Aircoookie, 2020-06-11:
Hi, yes, `lor` is used to toggle on an off the acceptance of E1.31 data.
WLED will always listen to MQTT, just that e.g. color changes will have no effect because the E1.31 data is displayed. With return to stream I mean that once `lor` is set to 0, the E1.31 data will be displayed again :)

pille, 2020-06-23:
how about introducing a (configurable) priority, so a user can define which protocol wins over others? it may be a top-down-list where higher positions override lower ones.

alborzs, 2020-06-23:
Sounds like a good idea as long as the priority can be changed on the fly, over mqtt or HTML API

drinfernoo, 2020-11-04:
This does appear to be implemented and working now, in case anyone couldn't figure it out... It's not in the docs 😅

drinfernoo, 2020-11-08:
I've added it to the docs apparently... expected it to ask me for a PR 😅 

https://github.com/Aircoookie/WLED/wiki/HTTP-request-API/_compare/41ae7f8fb0457bc357e9a8a9480b834259a7630b...885567599259c05bc9f5f12e9b5bddd8f0b8f2b1

closed 2022-10-21 by Aircoookie