wled/WLED #981: WLed - Wemos D1

MArb1985, 2020-06-09:
Hey, ich würde gerne meine WS2812b Stripes (193 LEDs) mit einem Wemos D1 Mini und externem Netzteil betreiben, funktionierte bis jetzt mit Tasmota und mit McLightning ohne Probleme.
Schließe ich auf D4 an, auch mit 3,3kOhm Pullup fährt der Wemos nicht mehr hoch.
Benutze ich RX fährt der Wemos hoch und ich komm in die Weboberfläche.
Habe dann einen 500 Ohm Poti zwischen die Datenleitung gelötet, bekomme ich ca über 100 Ohm die Meldung, LED Stripe nicht erreichbar, zwischen 80 und 100 Ohm Leuchten alle LEDs Orange, aber bootet ständig neu und darunter funktioniert es auch wieder nicht.
Habe schon mehrere Wemos probiert und die Datenleitung auf 20 cm gekürzt.
Gibt es da irgendwie eine Lösung?

mike811, 2020-06-11:
Hi,
ja einen anderen (besseren) esp8266 nehmen. Hatte mit den Wemos D1 (Clonen?) nur Probleme mit WLED. Seit dem Wechsel auf ein etwas teureres Board läuft es wesentlich stabiler und ohne Flackern.

mike2nl, 2020-06-11:
@MArb1985 
Esduerfte Dir bekannt sein das D4/GPIO2 nicht dafuer geignet ist. Hast Du dir mal die Belegung bei expressif angesehen?
Bei Tasmota geht das nur weil der GPIO2 ganz anders gehandhabt wird. Und ausserdem ist Tasmota eine art AUtomatisierungs Software, kein LED Animation Tool. AUsnahme ist der D4 wenn ein TFT benutzt wird dann ist das der TFT-CS. Und in Tasmota wird Dir jeder Admin abraten den GPIO2/D4 zu benutzen. Ich war 2 Jahre ADmin bei Tasmota bis es mir zuviel wurde um staendig alles zu wiederholen weil keine lesen wollte.

Hier mal eine Liste warum das nicht geht. Einfach mal lesen:
- D4	= IO, 10k pull-up, BUILTIN_LED = GPIO2
- Look ate the following image. Pins in yellow can't be used, or depending on the function you expect. So all yellow gpio's are no good choice at all.
[image]

MArb1985, 2020-06-11:
Ich habe die Version 3.1.0 von Lolin, der pin D4 ist Orginal und habe es auf den Rx abgeändert, funktioniert trotzdem nicht.
Hab auch ein Node MCU probiert, immer das gleiche.
Im Gegenzug dazu immer den gleichen Chip mit McLightning getestet um auszuschließen das etwas kaputt ist und da hat es funktioniert.

Legsmaniac, 2020-06-11:
Ich habe The Boss die gleiche Frage bezüglich der Verwendung von Pin 2 gestellt, da ich selbst Pin 5 oder 4 bevorzuge und weil ich häufig Bootprobleme bekomme, wenn ich NodeMCU 8266 und Pin 2 verwende. Seine Erklärung war, dass Pin 2 das Hardware-Fahren unterstützt, während andere Pins verwenden Software-Fahren, nämlich BitBang. Und dies kann zu Flackern und Instabilität führen. Sehen Sie, was sie in der NeoPixel-Bibliothek sagen ...

https://github.com/Makuna/NeoPixelBus/wiki/ESP8266-NeoMethods#neoesp8266bitbang800kbpsmethod

Model-Man, 2020-08-02:
With regards to flickering on the 8266 clones which often output lower levels on the data line I've found a simple solution. Using a single WS2812b pixel connect it directly to the output of your ESP 8266 (D4) via a very short lead. Power the single pixel via a full 5v feed. This is important as you want the full data output available from this pixel. Now connect the single pixels data line only to your main set.  Check ignore first pixel on Wled and your ready to go. 
Summery - The dedicated single pixel acts as basic amplifier so boosting the signal for you main pixels. With my test example I was able to run a data line of at least 3m without any flicker.

wilson3682, 2020-08-07:
Flickering solved with this fix:
https://github.com/Aircoookie/WLED/issues/230#issuecomment-670724521

closed 2020-12-13 by stale[bot]