wled/WLED #985: Transition lag when using slot 16 with multiple segments(3)

ToXIc-Dev, 2020-06-11:
Im using an ESP32 and the transition (power on/off, brightness) lags when using 3 segments on slot 16.

I could've sworn this didn't happen at first but its happening now and I haven't changed anything.

ToXIc-Dev, 2020-06-11:
I erased the flash using esptool and reflashed WLED and now it fades smoothly, no idea why it happened though.

ToXIc-Dev, 2020-06-11:
So I have found out that when the segments are saved to slot 16 is when the transition fade starts to happen.

ToXIc-Dev, 2020-06-11:
I tried two segments (each 30 LEDs) saved to slot 16 and it fades just fine

Aircoookie, 2020-06-12:
I'm having a bit of trouble visualizing the lag you are experiencing... Can you send a video of the LEDs when it occurs?

ToXIc-Dev, 2020-06-12:
So it appears to be only after saving 3 segments to slot 16 and then rebooting, then if I load slot 16 and turn it on/off or change brightness it happens, here is a video.

https://gfycat.com/finefearfulaxolotl

Aircoookie, 2020-06-12:
Thank you for getting back to me with a video so quickly!
Right after reboot a 2 second lag is normal while the WiFi initializes. But a lag after manually loading slot 16 is not something that should ever happen. It appears so from the video, but just to confirm, the segments are all on Solid, correct?

ToXIc-Dev, 2020-06-12:
Yes, they are on solid.

Aircoookie, 2020-06-13:
I believe I was able to reproduce this.
Had a hard time finding this bug, because it only occurs if both Segment 0 has been deleted and the segments are read from the preset slot! Should luckily be an easy fix though!

ToXIc-Dev, 2020-06-13:
> I believe I was able to reproduce this.
> Had a hard time finding this bug, because it only occurs if both Segment 0 has been deleted and the segments are read from the preset slot! Should luckily be an easy fix though!

Props for finding it so quickly, great work as usual.

Glad it's an easy fix and nothing to do with my esp. :)

Legsmaniac, 2020-06-13:
Wow!  Such dedication, persistence and patience to your software and your user base.  Something that is so very rare to be seen in Devs.  Many Devs would have just said "Nothing wrong with my code, it's your setup." and that would be the end of it.
Kudos Mr @Aircoookie Boss Man Sir!

[image]

Aircoookie, 2020-06-14:
@Legsmaniac thank you! I want everyone to have a great experience using WLED - If I'd just develop it for myself, I would not have put it up on GitHub 😂

commit 4f7dc5be (2020-06-22): Added Sunrise nightlight mode and more UI features

-   Added Sunrise nightlight mode
-   Added Chunchun effect
-   Added `LO` (live override) command to HTTP API
-   Added `mode` to `nl` object of JSON state API, deprecating `fade`
-   Added light color scheme support to web UI (click sun next to brightness slider)
-   Added option to hide labels in web UI (click flame icon next to intensity slider)
-   Added hex color input (click palette icon next to palette select) (resolves #506)
-   Added support for RGB sliders (need to set in localstorage)
-   Added support for custom background color  [... cut]
closed 2020-06-22 by Aircoookie