wled/WLED #986: Relay Module Low Level trigger

JerryLutor, 2020-06-13:
How to connect an inverted relay?

Changing the mode does not help.

The relay is this:
[image]

raoulteeuwen, 2020-06-13:
If i understand you correctly, i had a similar situation, and Aircookie explained how to solve and even supplied a firmware: https://github.com/Aircoookie/WLED/issues/631#issuecomment-605512524 . You can also order relays with a dip switch so you can change behaviour. Also see https://github.com/Aircoookie/WLED/wiki/Control-a-relay-with-WLED .

lancer199135, 2020-07-12:
I'm not sure if my relay is setup right. It turns on and off, but WLED still shows estimated usage around 250ma when off. Getting same reading on a strip with no relay.

raoulteeuwen, 2020-07-12:
Maybe share a picture showing the wiring? And what firmware are you running exactly? Do you have a multimeter?

lancer199135, 2020-07-12:
Not sure how to diagram like the ones I see online. I do have a meter, for some reason it will read volts in DC but amps it stays at AC. I've checked while it's off and it reads nothing. When on I'm getting about .1a. I have a logic level shifter thrown in but basically wiring is relay signal side going to esp8266 pin d6 with 5v power and ground commented. Power going to NO on relay and then out to the led strip. DR on ESP8266 to DIN on LED strip. The relay is turning on and off. I'm just wondering if the WLED app should be reading zero or not. On 0.10.0.

closed 2020-11-17 by stale[bot]