wled/WLED #988: Unwanted Strobe

Vliegtuigje16, 2020-06-13:
Hello

I’m building a light system with WLED but I’m having a weird problem. 
I’ve tried to find my problem on here but there’s just to much information and stuff, there for I’ve opened an new issue

The components I’m using:
* Full color 12v WS2811 (50 led’s)
* 12v, 6A power supply (JC power, JC-75-12) (also tried another one before)
* Power converter 12v -> 5v 50w (daygreen b10-1224-05) (tried 2 different ones)
* Nodemcu v3, ch340, with WLED (tried multiple, works great with an LED strip, but not whit this setup)

When testing it before, there was no problem and everything was working like you would expect. But now all leds are flickering like crazy when I’m turing them on, better called stroboscopic. This will happen in all kind of FX modes and also in steady. It will strobe in different colors, but it gets less when I’m approaching the data cable with my hand, then the flickering almost stops when I’m touching the data cable, and it works almost like it should. 

Does anyone know what’s going on? But more important, how can I fix this. 

Thanks. 

Regards, Robin

Aircoookie, 2020-06-14:
Hi Robin!

Most definitely your WS2811 require a higher data signal voltage than your NodeMCU can provide, meaning you can either add a single pixel of the strip that worked between ESP and WS2811, or use a levelshifter. See https://github.com/Aircoookie/WLED/wiki/FAQ#reason-2 :)

Vliegtuigje16, 2020-06-14:
Thanks for your answer, it’s weird because it worked before with the same setup. Same 50 led’s, same power supply, same power converter. Before no flickering, just steady and responsive to all FX modes. 
Even with 6 led’s connected, it won’t work properly now. I think it’s difficult to ad an 5v led pixel to the 12v string, so I’m trying the level shifter option. But still it’s weird that when I’m touching the data cable, it seems to work better. Like there’s some static interference or whatever.

Jonovich, 2020-06-17:
Try using a ferrite core [https://www.aliexpress.com/item/32627776378.html?spm=a2g0s.9042311.0.0.246e4c4dDaNCcI](url) on the data/power cables. Worked for me with the same symptoms you described.

Vliegtuigje16, 2020-06-17:
> Try using a ferrite core [https://www.aliexpress.com/item/32627776378.html?spm=a2g0s.9042311.0.0.246e4c4dDaNCcI](url) on the data/power cables. Worked for me with the same symptoms you described.

Thanks! I’m gonna order some, maybe I’ll have one laying around somewhere.

Jonovich, 2020-06-17:
I robbed one from a Garmin USB cable to test first. They just unclip. The data and earth cables, as they emerge from the NodeMCU, go through the ferrite, loop around the outside of it and back through a second time before heading to the WS2811's.

Vliegtuigje16, 2020-06-17:
I remember there was one inside of an old dvd recorder I’ve disassembled long time ago, only the cables are in a big box in my basement. Need to search for it. In the meantime I’ve ordered some of AliExpress, always comes in handy for future projects if it works

Vliegtuigje16, 2020-06-17:
> I robbed one from a Garmin USB cable to test first. They just unclip. The data and earth cables, as they emerge from the NodeMCU, go through the ferrite, loop around the outside of it and back through a second time before heading to the WS2811's.

I remember there was one inside of an old dvd recorder I’ve disassembled long time ago, only the cables are in a big box in my basement. Need to search for it. In the meantime I’ve ordered some of AliExpress, always comes in handy for future projects if it works

Vliegtuigje16, 2020-08-16:
> I robbed one from a Garmin USB cable to test first. They just unclip. The data and earth cables, as they emerge from the NodeMCU, go through the ferrite, loop around the outside of it and back through a second time before heading to the WS2811's.

Well, it took some time to receive the ferrite cores, unfortunatly it didn't work out, ive tried it on every cable, once, or in a loop, the 230 power cable, the 5v cables, 12v cables, +only, - only, only around the data cables, nothing worked... 
But still i think its really weird, because it worked without problems, after putting them in a plank with ping-pong balls around them they started to flickerd

Jonovich, 2020-08-16:
Doh - sorry - all I can say is that it worked for my setup that was a perfect equipment match for yours and did suffer the same symptoms prior to fitting the ferrite core. I too tried all sorts of things, including level converters prior to getting it working with the ferrite. Be very interested to hear what you resolve the problem to.

closed 2020-12-24 by stale[bot]