wled/WLED #990: Updateme Problem

klrock, 2020-06-13:
Last year I had flashed a Lolin nodemcu for WLED.  Now I want to update to new firmware.  When I power the module, an Updateme AP appears.  I can login using "updater1" but 192.168.4.1 does not appear.

I can get its settings as listed here:
SSID:	UpdateMe
Protocol:	802.11g
Security type:	WPA2-Personal
Network band:	2.4 GHz
Network channel:	7
Link-local IPv6 address:	fe80::559c:9957:feaf:a6e6%9
IPv6 DNS servers:	fec0:0:0:ffff::1%1
fec0:0:0:ffff::2%1
fec0:0:0:ffff::3%1
IPv4 address:	192.168.4.2
Manufacturer:	NETGEAR Inc.
Description:	NETGEAR A6100 WiFi Adapter
Driver version:	1030.3.831.2015

I tried to overflash the module using Nodemcu Flasher.  It appeared to be sucessful showing the scan code, Macs and completed flash progress.

However. When powered up the module still posts an Updateme AP.

I am out of ideas and hope you can steer me in the right direction.

Thanks,
Kerry

Aircoookie, 2020-06-14:
Hi Kerry,
did you over-flash the NodeMCU with the minimal http updater or a WLED binary? In some cases the WiFi information might not be updated correctly, meaning you could log into the actual WLED AP with the `UpdateMe` credentials. In that case `4.3.2.1` is the correct IP address :)

If you can't get it to connect at all, don't worry. You can just erase the entire flash once using esptool as described here https://github.com/Aircoookie/WLED/wiki/Install-WLED-binary

For future updates, you can use the `Manual OTA update` function in Security settings to upload a new binary. Hope you can get it to work :)

klrock, 2020-06-15:
Thanks for the timely advice.  I did have to erase it completely.  But it is restored to WLED now.

closed 2020-06-15 by Aircoookie