wled/WLED #991: WLED in Home Assistant

pepeEL, 2020-06-13:
Hi. Some Time ago i found WLED in integration of Home assistant. Now when try search in integration he not found me WLED. It was removed from Home assistant?

flbas1, 2020-06-13:
its there
https://www.home-assistant.io/integrations/wled/

it works nicely as long as they are both on the same network.  don't use the MQTT option since HA will pick it up automatically once the integration is added

TWitzenrath, 2020-09-13:
As noted, this does not work when Homeassistant is on a different network. This is also the case if homeassistant is running inside a docker container, which is quite a common setup these days. Would be nice if the security feature (allow only from local) could be made optional. Yes, I know the implications, yes I'm still sure I want this.

closed 2021-06-16 by stale[bot]