wled/WLED #994: I can't connect to the WiFi (Esp8266)

SchranzerTV, 2020-06-14:
Hi,
I have tried several esp8266.bin softwares .... none works for me ... I cannot connect to the Esp8266 (WLED-AP) wifi ... it always gives me an error when connecting ... I I flashed the software with the Esp-home-flasher .... I ask for help!
MfG Alex

raoulteeuwen, 2020-06-14:
Did you find the FAQ and did you try all that seems to apply? > https://github.com/Aircoookie/WLED/wiki/FAQ . If so, please follow https://github.com/Aircoookie/WLED/wiki/FAQ#if-i-report-a-new-issue-what-do-i-need-to-share and provide some more info.

SchranzerTV, 2020-06-14:
The whole thing has worked like this before! I took another esp8266 because I wanted to do it again, but it just doesn't work. I cannot connect to his wifi no matter what I do .... I have the Esp8266

SchranzerTV, 2020-06-14:
Ok .... the esp8266 was carp

closed 2020-06-14 by SchranzerTV