wled/WLED #996: Multiple Segments causing blinking pixels?

CMaxim77, 2020-06-16:
Hello, thanks so very much for sharing your work. Because of your awesome software I finally started making led projects in my home and for friends. I appreciate the work you put into this!

This is more of a question then a issue I believe.
I modified my address marker on my house to allow led strips to light up behind my house address. It looks fantastic and is very visible now at night.

If I use all 36 pixels as one segment everything works perfectly during animations, but if I split up the pixels in 2 segments (18) each I get random blinking pixels. Do you think it needs more power, or does it require more to run multiple segments?

I’m just trying to figure it out so I know in the future if this comes up again.

Thanks again for your time!!

Aircoookie, 2020-06-16:
Hi! I'm glad that you like WLED :)

Do you have only two Segments (`0 & 1`), the first set to LEDs 0-18 and the second to 18-36.
Some users have a third segments set, e.g. leave Segment 0 as covering all LEDs. Segments should not overlap, that can cause flickering in the overlapping LEDs.

If you're segments do not overlap, the flicker most likely stems from the data signal voltage being to low. In that case you'd need a sacrificial LED or levelshifter:
https://github.com/Aircoookie/WLED/wiki/FAQ#reason-2
But in that case you should usually get flickering also with a single segment.
Does this happen on all or a specific effect?

Morphy99, 2020-06-17:
I'm playing around with segments with two vertical strips connected to one ESP too. I'm slightly confused with a few things and I have noticed the same behaviour as above. I initially had two segments 0 and 1 (0-30 & 30-60). After sending wled.effect from HA to it I noticed only segment 0 was controlled so I added another segment 2 and changed 0 to be 0-60. This is where things started to get weird. If I have all three segments ticked and I change the effect from solid I get the effect on seg 1, seg 2 does something weird like the effect but a lot slower and sometimes seg 1 starts flashing with the solid colour in between the effect. 
You mention segments should not overlap but how do you do this with 3 segments and one being the whole strip? 
Is it possible to send effects to both segments as if they were independent strips while also being able to control both strips as one from Home Assistant?

CMaxim77, 2020-06-17:
I did leave segment 0 as default and added 1 as 0-18 and 2 as 18-36 so that wasn’t good. I did notice after skipping the 1st pixel the issue seemed to stop. So I will try to add a additional pixel to the 8266 board as used in the example it looks like I just have to run it off of the 5v pin.

Seriously, I really appreciate the reply and love your work. I can’t wait to see what happens with future releases!

closed 2020-10-23 by stale[bot]