wled/WLED #997: How to sync everything, excluding one controller?

BezziHD, 2020-06-16:
Hi there,
I build together with my friend a Laser-sword. The room is full of WLED-controllers and the Laser-sword should be the only thing which doesn´t sync with the other controllers.
I want to use the Android-App for everything... But if I press the sync button everything (of course) works together.
Is there any way to sync everything, excluding one controller?

raoulteeuwen, 2020-06-16:
It seems you could put devices/strips you want to separate, separated by using a different UDP port. Not sure how good your German is, but afaik, this is what's talked about in https://github.com/Aircoookie/WLED/issues/890

BezziHD, 2020-06-16:
Ok Dankeschön :) Komme aus Deutschland

closed 2020-06-16 by BezziHD