wled/WLED #1001: WPA2-Enterprise support?

EssKayz, 2020-06-17:
Hi,

sounds like your company network is an enterprise network. Those aren't easy to connect to, currently WLED only supports home networks (WPA2-PSK). It seems like it is possible though: https://www.google.com/search?q=esp8266+enterprise+wifi

I'll keep this in mind and maybe I can add support soon!
Cheers!

_Originally posted by @Aircoookie in https://github.com/Aircoookie/WLED/issues/242#issuecomment-537598332_

Hi!
Is there currently any plans for WPA2-Enterprise to be supported?
If not, I'll see if I can't jerry-rig it in my local copy somehow :)

garyd9, 2020-07-11:
I'll just leave this little block of code I inserted in initConnection() here (mostly taken from various google search results)...  Oh, and it does work.  On my particular network, my username must be "domain\\user" instead of just "user", but it does work fine on PEAP/MSCHAPv2 (coming from an ESP32 device)

```
  WiFi.disconnect(true);
  WiFi.mode(WIFI_STA);
  esp_wifi_sta_wpa2_ent_set_username((uint8_t *)username, strlen(username));
  esp_wifi_sta_wpa2_ent_set_password((uint8_t *)password, strlen(password));
  esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT();
  esp_err_t code = esp_wifi_sta_wpa2_ent_enable(&config);
  Serial.print ("enable response code: ");
  Serial.println(code);
  const char *ssid = clientSSID;
  Serial.print("wifi Begin connection to: "); Serial.println(ssid);
  WiFi.begin(ssid);

```

EssKayz, 2020-07-14:
I eventually did manage to get it working aswell, I just had to give up on using ESP8266, and buy an ESP32. Also just ended up coding everything myself from scratch, since I decided I'll save myself some headache. Not sure if ESP8266 can even be made to work with all WPA2-Enterprise networks :)

closed 2020-11-19 by stale[bot]

tudi235, 2022-03-06:
Hey! I didn't managed to get it working. Can you copy paste your code under initConnection? Thanks :)