wled/WLED #998: Changing the number of channels - E1.31 vs ArtNet

pk88888, 2020-06-17:
Hi!

I have a problem with the operation of the device when I choose more than one universe of pixels. When the E1.31 mode is selected, everything works fine, but the change to ArtNet causes a conflict due to the wrong number of channels.

I would like to use the Artnet multicast function because it gives the best results and the best performance.

By default, the universe has 512 channels, but it is often practiced to change to 510 due to the simpler division for 3 channel pixels.

Is it possible to change the number of channels / universe in the software?

Aircoookie, 2020-06-17:
Hi!

> By default, the universe has 512 channels, but it is often practiced to change to 510 due to the simpler division for 3 channel pixels.

This is exactly what we are doing in WLED! I could add support for the full 512 channels, but it would make the receiving code more complicated. Which sending software causes this conflict and how does it affect the data sent? Do the pixels of the second universe have the wrong color?

pk88888, 2020-06-18:
Thank you for such information!

The problem appears when I send a given animation to the leds and switch between artnet and sacn. When working on sacn, everything works fine and when switching to artnet, colors at the end of the universe begin to mix. The first thing that occurred to me was that there are 510 channels in sacn and 512 in artnet in WLED. I can record a video if it helps.

pk88888, 2022-01-12:
I have now installed the latest version of the software and the problem with the different colors on the connection no longer occurs (when using artnet). At the moment there is a problem with a blue flashing pixel at the point of the next universe. 

Has anyone found a solution for this? 

I found a similar topic and it is clear that this problem is due to the fact that most programs send 512 channels according to the standard and not 510.