wled/WLED #999: Web UI Customization Issue

adamw86, 2020-06-17:
First I just wanted to say thanks for making such a great library and UI!

My issue is the compression process in the npm scripts for the index.html --> html_ui.h.  I have a background image linked in the web UI that worked fine with 0.91.b version.  

     background-image: url('http://imageURL/');

It appears that when I leave that image link in the html file, the compression script is linking actual image file into the C array output to html_ui.h.  It works fine with the line of code removed, but grows by many MBs when the img link remains:

Inlined 4880446 characters
Compressed 3602762 bytes
Writing wled00/html_ui.h

Any ideas how to resolve this?

Thanks!

Aircoookie, 2020-06-17:
Hi!
We best ask @huksley ! I wouldn't expect the script to actually go ahead, fetch that resource and integrate it into the C array. It might be a cool feature for including JS libraries automatically, but yeah, in this case it becomes an issue.

Apart from the root issue, interesting that you would think of putting a background image at the same time as me, as this is something I'm actually currently adding support for! Here is a little preview 👍 

[image]

Here is the new index file for you to test if you'd like:
[index2006170.zip](https://github.com/Aircoookie/WLED/files/4793947/index2006170.zip)
The link to the image is saved in Chrome local storage, so it should work around this automatic inclusion issue. I would say only files up to say 50kB should be integrated into the array, but I don't know whether that is actually possible in the script.

huksley, 2020-06-17:
Hello @adamw86 
could you rewrite tools/cdata.js 
and change this line in function writeHtmlGzipped
`new inliner(sourceFile, function (error, html) {`
to this line
`new inliner(sourceFile, { images: false }, (error, html) => {`
the idea is to disable all image embedding.

But I would not suggest to do this for the master branch of WLED.

The idea is to have all resources packed in the firmware, not referenced from some external (potentially unsafe) source.

closed 2020-10-23 by stale[bot]