wled/WLED #1003: Which GPIO to use with WS2815B

MC-1984, 2020-06-18:
I just cannot find anywhere in the Wiki how to connect a WS2815B to an ESP32. I have several WS2812 which use G2. But the WS2815 has 2 data lines (BO/DO) instead of just one. How do I connect the additional data line?

Aircoookie, 2020-06-19:
Hi! According to the datasheet, you should be able to connect DI to gpio2 normally. The BI of the first pixel should be connected to ground!

MC-1984, 2020-06-22:
Thanks for your answer. Unfortunately, connecting BI to ground resulted in some LEDs just lighting up even if the ESP was disconnected from power. There is surprisingly little information about that topic on the internet. So if anyone stumbles across this threat: You do not have to connect BI at all if the first LED is working. The first LED that is connected to DI will send the signal to DI and BI of the second LED.

closed 2020-06-22 by MC-1984