wled/WLED #1005: Light sensor module

costads, 2020-06-20:
Hi guys,

I read on another forum about a guy suggesting to use a light sensor module with digital output ( https://www.aliexpress.com/item/32571120284.html ) to make my LED strip turn on at night, and off during the day.

I have connected the module to D3, setup the macro T=2, and as such is working, however, the light sensor has to reach on/off/on state before it will turn off the light, and again off/on/off to turn it on again.

Is it not possible to make it work like I want to somehow?

Aircoookie, 2020-06-20:
Hi!
Please do not post the same both to issues and the forum. I replied to you there :)

closed 2020-06-20 by Aircoookie

costads, 2020-06-20:
Sorry, didn't see you had the forum untill today :)