wled/WLED #1008: No WLED AP - Multiple Boards - Wemos Mini D1 Pro

robotica72, 2020-06-22:
Hoping for a suggestion - Long time user of WLED - So first, thanks so much for the project!

Is there anything I can to do to figure out why a new Batch of Wemos Mini D1 Pro's refuse to show the WLED AP after a flash?   Im just flashing the release binary of 0.10 using the ESP flasher tool.  Just for my own sanity, I flashed a board that I bought months ago and had no issues.  In my current batch of 10 Wemos boards, I have flashed 3 and none of them show the WLED AP.

I have also tried, basically, every flash in the latest release - none of them show the WLED AP...

Any ideas?

raoulteeuwen, 2020-06-22:
Did you try 0.8.5 and or releases after that one also and did none of those work? If one of those works, try OTA update to get to 0.10.0?

Aircoookie, 2020-06-22:
It might also be that the particular batch is pre-loaded with a firmware or bootloader incompatible with WLED. Try esptool as the flasher, `esptool.py erase_flash` followed by a binary flash will most likely solve the issue!
Also, are you trying to boot them without LEDs connected? That is usually a good idea to make sure it is not a hardware issue :)

robotica72, 2020-06-22:
@raoulteeuwen - Yeah, I tried 0.8.5, and a few other versions - all of them have the same issue.

@Aircoookie - Just tried that and same results using esptool to erase and flash.

So, since everything seems to be failing, I decided to flash the basic WiFi examples for the ESP and it seems that nothing related to WiFi works at all.  I tried the basic client, http server, etc...  Its almost as though this batch of Wemos has WiFi issues.   As a last ditch effort, Im going to enable the external antenna and see if that fixes anything.  Im hoping the issue may be just bad built-in antennas - I use external antennas for my projects, so this wouldnt be a big issue for me.

Something tells me thats not the root cause - but Im grasping at straws now...

robotica72, 2020-06-22:
Ok, Antenna didnt fix it - but I did flash the WiFi scanning sketch from the examples and it does see different SSID's - So the Chip and WiFi should be all good.   Of course, I think I would have preferred that the scan showed nothing - would point to a root cause at least :)

war6000, 2020-07-06:
having the same issue, tried using the bin with several tools before and after `esptool.py erase_flash` and several firmwares too, the only one working for me is 0.8.5 also tried doing OTA from that version to 0.10 and i'm not able to access it after it, only flashing back to 0.8.5 works. Next step would be to compile it my own since any bin file >0.8.5 is not working. (nodemcu4MBytes)

robotica72, 2020-07-07:
So, with this set one set of boards, I am 0/10 using any Firmware version.  Even trying the WiFi examples, only the scanner works - Any other connection fails - it doesnt work as a client, web server, AP, nothing ...   I forgot I had  previous batch of 10 Mini Pros from a different seller and I flashed the .10 firmware to three of them without any issues - WLED-AP came right up and I added them to my LED network (VLAN).  So it looks like, no surprise, that some ESP's from sellers are the issue.  Should have guessed when they were advertised as 16Gb and they were are actually 4Gb.   They work with all other code, so they must have some WiFi issue - for some reason.

If anyone on the main project team wants me to send them one for any testing, let me know - Happy to send one over to see if you can figure anything out for future questions from other people.  (Prefer in the US due to Shipping)

closed 2020-11-12 by stale[bot]