wled/WLED #1009: Button press syncing

whitey2861, 2020-06-24:
Hello, I just wanted to say that i love this project. After spending far too long coding arduino minis (not my strong point) this is such a great UI. 

 I am having a little issue with my current project, I have 2x D1 minis driving led strips. I have them setup and synced fine. I have a button setup on each of the D1's and have the macro assigned to cycle between different effects. All good up to this point.  My question is : is there a way to make these buttons sync with each other. Right now if I press one button it only changes the effect on the board that it is connected to ( even with the boards synced). 

My end goal is to be able to switch between effects but pressing either of the buttons and for both strips to update. 

Any help would be appreciated. 

Thankyou

Aircoookie, 2020-06-24:
Hi!

This should be quite easily possible, just enable `Send Macro notifications` in Sync settings :)

whitey2861, 2020-06-24:
You legend, Thank you !  I completely missed the sync setting tab!

closed 2020-06-24 by whitey2861