wled/WLED #1011: Udp reatime to LEDs group

tihoangyeudau, 2020-06-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I wanna wled received data transmite to LED groups.

**Describe the solution you'd like**
When i use ambilight vie udp, i want each one pixel led from ambilight send to wled for each group of 9 leds.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for your ideas for making WLED better!

Aircoookie, 2020-06-26:
This is a good idea! I will consider making realtime (UDP, E1.31., ...) work with Segments. That would allow you to choose offsets and grouping, and even allow for running WLED effects simultaneously in a different Segment. I will take a look on how to add this!

Aircoookie, 2020-06-26:
For now, you could hack this in into `setRealtimePixel()` in `udp.cpp` by the way!

```cpp
void setRealtimePixel(uint16_t i, byte r, byte g, byte b, byte w)
{
  i = (i + arlsOffset)*9;

	if (!arlsDisableGammaCorrection && strip.gammaCorrectCol)
	{
		for (uint16_t pix = i; pix < i + 9; pix++)
		strip.setPixelColor(pix, strip.gamma8(r), strip.gamma8(g), strip.gamma8(b), strip.gamma8(w));
	} else {
		for (uint16_t pix = i; pix < i + 9; pix++)
		strip.setPixelColor(pix, r, g, b, w);
	}
}
```

tihoangyeudau, 2020-06-26:
Thank you for your try. I will test it tonight

tihoangyeudau, 2020-06-26:
> For now, you could hack this in into `setRealtimePixel()` in `udp.cpp` by the way!
> 
> ```c++
> void setRealtimePixel(uint16_t i, byte r, byte g, byte b, byte w)
> {
>   i = (i + arlsOffset)*9;
> 
> 	if (!arlsDisableGammaCorrection && strip.gammaCorrectCol)
> 	{
> 		for (uint16_t pix = i; pix < i + 9; pix++)
> 		strip.setPixelColor(pix, strip.gamma8(r), strip.gamma8(g), strip.gamma8(b), strip.gamma8(w));
> 	} else {
> 		for (uint16_t pix = i; pix < i + 9; pix++)
> 		strip.setPixelColor(pix, r, g, b, w);
> 	}
> }
> ```

it works perfectly. I love it. You can add segments and group led base on this function.

ReviloEgros, 2021-12-29:
Any news on this if it will be implememted? A feature i would like to see too!

blazoncek, 2022-10-10:
Displaying realtime (live) LED data on main segment (leaving other segments to be used by WLED) available in 0.14

closed 2022-10-10 by blazoncek