wled/WLED #1012: WLED Alexa Device, RGBWW SK6812, white channel problem

TechnoHoschi, 2020-06-26:
Hello,

I control my SK6812 RGBWW using the Alexa device. red, green, pink, everything works perfectly. however, if I switch to white once via Alexa, the following colors are shown with too high white channel. Only after I manually set the white channel in the web interface to 0, the colors fits again. However, only until the next time Alexa is switched to white.

I use version 0.10.1 on an ESP32, tests on an 8266 were identical. The versions below 0.9.1 work correctly.

regards

michael

Aircoookie, 2020-06-26:
Hi,

thank you for the report!
Indeed this is the case, the white channel is not reset to 0 when selecting non-white. Will fix!

TechnoHoschi, 2020-06-26:
Great,

thank you.

Aircoookie, 2020-06-26:
Done!

commit 5464cbc2 (2020-06-26): Fix Alexa not turning off white channel (fixes #1012)
closed 2020-06-26 by Aircoookie