wled/WLED #1014: Wemos D1 mini  - Error 2 -

djhampson, 2020-06-27:
I get an "WLED Error 2" on 2 Wemos D1 Minis. I have tried 2 different releases and am definitely not using the 1M binary. 

**WLED version**
Board: Wemos D1 mini x 2
Releases used:
WLED_0.10.0_ESP8266.bin
WLED_0.9.1_ESP8266.bin

My only suspicion is that my devices are not true Wemos d1 minis and they are some sort of knock off (they were purchased from ebay). i can run Tasmota on them without any issue. 

[image]
[image]
[image]

Thanks in advance for your help - let me know if i can provide more info.

raoulteeuwen, 2020-06-27:
I assume you found https://github.com/Aircoookie/WLED/issues/900 . You did follow https://github.com/Aircoookie/WLED/wiki/Install-WLED-binary ? Maybe try https://arduino.stackexchange.com/a/24692 to see what info that gives?

djhampson, 2020-06-28:
Thanks for the reply @raoulteeuwen 

I did follow https://github.com/Aircoookie/WLED/wiki/Install-WLED-binary however I deviated slightly by using Tasmotizer (which I understood used esptool) to upload the binary.

When I used ESP Home Flasher it is working fine. 

So learning is - if you get the WLED Error 2, firstly check you are using right binary https://github.com/Aircoookie/WLED/issues/900 and then check the flashing method you are using.

closed 2020-06-28 by djhampson

newwolf, 2020-09-30:
Experienced the same issue with my Wemos D1 mini and WS2801.
Flashing "WLED_0.10.2_ESP8266_ws2801.bin" with Tasmotizer gave me Error 2 and loosing Wifi Info.
But flashing "WLED_0.10.2_ESP8266_ws2801.bin" with "ESP8266Flasher" on 0x00000 works perfect and without errors.