wled/WLED #1016: Add brightness ramping for button long press

Aircoookie, 2020-06-28:
**Is your feature request related to a problem? Please describe.**
Brightness setting not possible without remote control

**Describe the solution you'd like**
Support brightness ramping (gradually brighter, then off and repeat) for button long press

**Additional context**
By Sam on Discord

ToXIc-Dev, 2020-06-29:
I've had a idea that will be better than gradually brighter, then off and repeat.

So when you hold it, it ramps up to full and when you get to full it stops, you then have to let go of the button and hold it again for it to go down and then stop at lowest brightness.

This makes more sense otherwise it will be a pain to get it at full brightness, its more intuitive IMO.

Aircoookie, 2020-06-29:
That is actually an excellent idea! I have had another problem with allowing the user to hold the button for longer than a few seconds, which is that it would interfere with the "press button for 7sec to open recovery AP" functionality, which would also be solved by your proposal. Thanks!

jdtsmith, 2021-02-19:
If the numerical increment with number would wrap around you could do something like this now with say double click attached to A=~64.  Not sure why that doesn’t wrap.   Maybe we could allow an optional “w” to indicate numerical wrap around, like A=~64w?

Lenbok, 2021-03-04:
I'm looking at whether I can convert my custom light firmware to WLED. Here's how I've implemented this feature in mine, and I quite like it in practice.

Short press toggles between off and maximum brightness. This also sets a "current fade direction" to either increase or decrease, respectively.

Long press ramps the brightness in whatever the current fade direction is, while held. Upon release, it also toggles the fade direction, allowing you to make additional fine adjustments. Long press never wraps around between min and max, as that would be quite visually jarring (if you want visually jarring, use the short press).

h07d0q, 2022-08-11:
I'd like to have this feature on segment basis!
I'm trying to setup 3 individual segments:
ID 0 LED 0 - 180 SK6812 RGBW
ID 1 LED 180 PWM white
ID 2 LED 181 PWM white 

Segment 0 is controlled via playlists.
Segment 1 and 2 are two areas in one room and each controlled via GPIOs:
Single click: toggle segment
Double click and hold: dim segment brightness (not main brightness because of seg0)
Triple click: reset segment brightness to 100%

My attempt for double click is not working:
`[{"id":1,"on":true,"bri":"~-10"}]}`

evazzoler, 2022-08-20:
I think this feature is very important because it is the missing part that transforms the software button in a real life button. Not me, but my wife and all other "normal" people expects to dim with long press and toggle on/off with short press. A different behavior is confusing the end user.
I think a led strip used in a home should appear as a normal lamp first, with standard controls, then you can turn on special effects and let the surprise exploding.
Please, consider this for managing the enhancement priority. I think is very high, and maybe it's not simply my point of view. Thank you!

blazoncek, 2022-08-20:
Long press brightness change is implemented on button 1.

evazzoler, 2022-08-20:
> Long press brightness change is implemented on button 1.

Ops... I missed that information! Where can I find more informations about button 1?

ToXIc-Dev, 2022-08-20:
> I've had a idea that will be better than gradually brighter, then off and repeat.
> 
> So when you hold it, it ramps up to max and when it gets to max it stops, you then have to let go of the button and hold it again for it to go down and then stop at min brightness.
> 
> This makes more sense otherwise it will be a pain to get it at min/max brightness, its more intuitive IMO.

@blazoncek ah yes, I see, nice, ideally it would work like this tho

ToXIc-Dev, 2022-08-20:
> > Long press brightness change is implemented on button 1.
> 
> Ops... I missed that information! Where can I find more informations about button 1?

It's the default button 1 long press action, so just change the button you use to button 1

blazoncek, 2022-08-21:
@ToXIc-Dev agreed, perhaps in next iteration.