wled/WLED #1018: Hart rate monitoring through Wled

PatrikPatrik2, 2020-06-30:
By using an ESP32 it is possible to get the hart rate reading from bluethooth chestband for hart rate monitor (like Polar) and send the hart rate data (for example: 165) to MQTT. This system I have already up and running.

In my gym I have a LED ligth strip controlled by WLED. Would it be possible to have WLED to subscribe to the hart rate MQTT topic and display different led pattern depending on the hart rate or changes in hart rate etc. (for instance flashing red when the hart rate is close to a set max pulse). Such a feature would make my excersize more enjoyable!

Aircoookie, 2020-07-02:
Hi, yes, this would of course be possible!
This would be a very cool idea for a usermod if you or someone else are looking for a nice project :)

Lasermatrix, 2020-07-10:
I really like this idea and I've been attempting to do this, I've got a Polar H10 and with code from the internet have managed to get readings from the HR monitor.

Sadly when trying to incorporate this as a usermod, it's blowing the program storage limit out of the water, even with disabling most of the modules, it's still running at 126% of total storage :(

My code is quite small, so it's got to be the BLE library.  Still keeping at it tho :)

Lasermatrix, 2020-07-10:
Actually, your idea is better!  A simple MQTT instance serving the data and WLED consuming it.  That's the way to go!