wled/WLED #1019: Project Aurora Integration

ikrowni, 2020-07-01:
**Is your feature request related to a problem? Please describe.**

With project aurora we could sync the lights from wled with the games currently playing on a PC

**Describe the solution you'd like**

Aurora integration/sync option

**Describe alternatives you've considered**

None that i can find

**Additional context**

Seems they are actively looking for more integration options to work with an ever expanding lighting setup

ToXIc-Dev, 2020-07-01:
Ooo yes, this would be great.

ToXIc-Dev, 2020-07-02:
Might be best to ask the project Aurora to support WLED instead tbh though as it likely won't require changes to WLED (could just use the HTTP API)

It would need to be added to project Aurora instead.

ikrowni, 2020-07-02:
I've already created a support ticket for it over there.

https://github.com/antonpup/Aurora/issues/2027

Aircoookie, 2020-07-02:
Yes, an integration would definitely be super cool! I'd suggest using the [JSON API](https://github.com/Aircoookie/WLED/wiki/JSON-API) primarily :)

ikrowni, 2020-07-20:
> Yes, an integration would definitely be super cool! I'd suggest using the [JSON API](https://github.com/Aircoookie/WLED/wiki/JSON-API) primarily :)

I was just informed by another user that they wrote a UDP script for aurora. After implementing the UDP script as per the instructions ive added my wled instance into aurora but it doesnt seem to be working. Im not sure if this is a problem with wled or the script. I have 644 leds to drive and i see that udp can only do a max of 255. Maybe you could modify the script to work better with wled? I hate to ask for you to implement something into someone elses software but this would be an amazing wled feature to have game reactive lighting.

Heres the link to the response i was given about the script im attempting to use 
https://github.com/antonpup/Aurora/issues/2027#issuecomment-660896679

Aircoookie, 2020-07-21:
The protocol they implemented is the Hyperion type 0 UDP protocol. It is supported by WLED, but you'll have to set the port 19446 in the integration. Unfortunately due to the maximum size of an UDP packet on the ESP platform, the maximum amount of LEDs supported is 490.

ikrowni, 2020-07-23:
>  the maximum amount of LEDs supported is 490.

Is there any getting around this at all? This is working great but the only problem is my led strip is only about 3/4 lit up.

closed 2022-10-21 by Aircoookie