wled/WLED #1020: Wiki Update to Setup with ARLS

jccroft1, 2020-07-01:
As stated on the below section of the Wiki, WLED is now supported by ARLS. 
[Setup with ARLS](https://github.com/Aircoookie/WLED/wiki/UDP-Realtime-Control#setup-with-arls)

The forked changes to fix the message to match the protocol (m.Append(1)...) and to support python 2 have been merged now though. 

Thus, this section can be updated as the forked version is no longer required, just use the latest. However, the UDP_PORT in config.py still needs to be changed as it defaults to 777 instead of WLED's default of 21324.

Aircoookie, 2020-07-02:
Hi and thank you for the heads up!
I couldn't find the protocol changes `m.append(1)` in the master https://github.com/scottlawsonbc/audio-reactive-led-strip , just the python 2 support. In fact, changing this would break support with their own ESP8266 sketch, so I thought they may have added a new device type for WLED, but that doesn't appear to be the case either...

jccroft1, 2020-07-03:
Apologies, you're correct, only the Python 2 changes have made it in.

Aircoookie, 2020-07-03:
Thanks for confirming! I removed the `Use python 3` instruction from the wiki as it now doesn't matter but kept the rest.

closed 2020-07-03 by Aircoookie

tchristle, 2021-02-07:
I just cloned the fork for ARLS support and I am able to get the visualization.py to run, but my light strip is not reacting. Did something else change that broke this?