wled/WLED #1021: Philips Hue Sync

renanbarbalho, 2020-07-02:
hello,
I configure my wled with ip of my hue brigde and i received this error:

[image]

how to solved?

Aircoookie, 2020-07-02:
Hi! It says that the light with ID `10` you are trying to sync to does not exist.
Please go to the About section of the Hue app and confirm that the light with the number 10 exists and is the one you want to sync to! If the light actually exists, I need to look into it because it might be bug.
Thank you!

renanbarbalho, 2020-07-02:
Thanks for your answer. Where I find the IDS? Because I have four lamps (sala 1, sala 2, game 1 and game 2) My first tried was set ID 1 with same error.
Do you have any idea?

toto79, 2020-07-08:
@renanbarbalho: there are different opportunities. Most of the time i use WinHue to explore my hue bridge. Under view -> bulbs view you'll see the Light IDs. It's also possible to do it with the hue api. ...

renanbarbalho, 2020-07-08:
@toto79 thanks for your help, this app was killer! I found the bulbs ID and now myLed was connected with my bridge.

Thanks a lot!

closed 2020-07-08 by renanbarbalho