wled/WLED #1022: Modular WLED ?

Legsmaniac, 2020-07-02:
**Is your feature request related to a problem? Please describe.**
I've noticed you've received a LOT of requests for enhancement, improvement and modification recently.  All great ideas however, I'm getting kinda concerned that the more modifications added over time, the bigger the program is becoming and in turn there's going to be more memory issues and devices/modules/nodemcu's no longer able to be considered for use due to future incompatibility.

**Describe the solution you'd like**
I'm wondering if one solution could be a Modular WLED?  A WLED with Extensions, like those of Chrome WebBrowser?  So those who just want the straight no-frills vanilla WLED for Christmas Lights, they simply need WLED base program. And those who want to use *_this mod_* they download and install the extension for *_this mod_* - and so on.

**Describe alternatives you've considered**
An alternative would be to offer just a basic WLED download for those who  just want the straight no-frills vanilla WLED for Christmas Lights?

Aircoookie, 2020-07-03:
Yes, I agree! Having WLED be more modular would allow you to only install the features you need and would also make it easier to e.g. port it to other platforms in the future.
We already have a sort of modularity, with the defines for disabling some features (MQTT, Alexa, huesync...) and of course usermod for dynamically adding sensors and other features. I would like to expand on the v2 usermod system soon, to make it possible to add for example UI components or effects via it.
Unfortunately, we will never reach Google Chrome Extension levels of convenience, because the C++ language requires us to recompile every time the code changes in contrast to Javascript or Python. So there will be no runtime "downloadable" plugins or anything like that unfortunately. But I would still like more flexibility with what functionality gets included at compile time!

Marlor, 2020-07-04:
It would probably be easiest to use one combined "config.h" for every option the end user can change like
[Tasmota](https://github.com/arendst/Tasmota/blob/development/tasmota/my_user_config.h) and [Marlin](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration.h)

huggy-d1, 2020-07-06:
Would be fantastic if there were a way to build WLED online with a build GUI.  Similar to this: https://github.com/benzino77/tasmocompiler where it allows compiling on gitpod.io (online).

Aircoookie, 2020-07-07:
More modularity: Good.
A config.h with everything in one place instead of the current mixture of wled.h, const.h and NbpWrapper.h : Also good.
An online compiler literally easy enough to use for anyone who found this Github: the best!
We need TasmoCompiler for WLED 😄

Marlor, 2020-07-11:
The merged config.h sounds like a good little projekt to lern github and pull requests.
I would like to try making one if you don't already have something planned.