wled/WLED #1025: ESP disconnects from wifi and/or restarts itself in a short period of time

golddragon007, 2020-07-04:
**Describe the bug**
I have multiple (8) magic home spi led controller that I hacked successfully. When I used WLED in AP mode it worked fine, I was able to control the LEDs, set animations, etc. But after I've set up my WIFI SSID name and password, I can't use it anymore. From the ubiquity control panel, I see that it connects and then 1-3 sec after it disconnects. Visually the following happens:
[image]
I tried to set up this in different settings, like the wifi keeps alive, turn of Alexa, etc. And currently, 4 out of 8 behaves in the same way, even with these different settings.

**To Reproduce**
I guess hack it and set the WIFI.

**Expected behavior**
Connect to WIFI and stay on it.

**WLED version**
 - Board: Magic home SPI LED controller
 - Version WLED_0.10.0_ESP8266_1M_full
 - Format binary

**Additional context**
Any idea?

zineer, 2020-07-04:
I'm experiencing what appears to be the same issue, but on a Wemos D1 mini. I've had the same D1 mini in use for awhile (WLED 0.9.1) and was away for a week, and when I came back Home Assistant was showing the WLED as unavailable. I reset the board and ended up with a reboot loop/flash as golddragon007 shows. I reset the board, did a data wipe and installed 0.10.0, but issue remains - once home wifi settings are entered, it does a reboot loop (blue light flashes on board every ~6.5 seconds, connected lights flash as well).

Aircoookie, 2020-07-05:
Hi,
sorry for the bootloop!
Try disabling the Alexa emulation in Sync settings before entering the home network info. It is known to fix the issue for the 1M binary in most cases although we are not yet aware what the root cause is. I hope it will help you!

golddragon007, 2020-07-05:
Hi
I've already tried it and didn't work. One of my 4 tries was like that.

Aircoookie, 2020-07-05:
Hi! Oh sorry, I clearly can't read. It's late here 😄 
One more thing you could try if you haven't already is also clearing the mDNS name in WiFi config (after turning off alexa).
If you do not rely on the interfaces provided by the 1M_full binary, you can also try the 1M_ota, which tends to be more stable.
Hope you will get it up and running soon!

golddragon007, 2020-07-06:
Hi @Aircoookie,
OK, Thanks, this mDNS actually did the 'trick'.
The experience after this, I would say you need more try-catch blocks. Prevent the app to crash. Even create a 'Log' menu where you dump out the last X crashes (backtrace). If you relying on a service, then make sure you catch the errors, even with some kind of generic way to prevent app crash or reboot loop. If you caught the error, you can simply log it somewhere and then the user can show it to you which will give more clue what's the problem with it.

Aircoookie, 2020-07-07:
@golddragon007 We do not have the luxury of exceptions on the 8266 without a significant performance overhead afaik :)
Logging is also not yet implemented (filesystem). However something I would like to add is an automatic "safe mode" to recover from conditions that cause bootloops (99% it is either a problem during connection to the home network or a memory exhaustion). That will save the inconvenience of a reflash. Will open a FR issue for tracking it!

daredoes, 2020-10-03:
I have been having this issue with an ESP32 flashed with WLED, and have most likely just solved it by flashing the software with ESPFlasher instead of ESPTool

daredoes, 2020-10-03:
I was wrong. Worked great for several hours, now having issues. Planning to dump the logs if any

daredoes, 2020-11-30:
So I was not entirely wrong. It was the result of a bad flash. I have a controller that 95% of the flashes I've done on it result in this issue in some form. If not immediately obvious, it comes out hours later. 

I had one flash where it "just worked" as expected. No more reboots or freezes. I tried using some other firmware, and bam I was right back to square one. 

I left my light disabled for a few weeks, and then decided to give it One Last Shot™ before the weekend. My board is being connected to my flashing board directly to some pins, so I held the connection together as firmly as I could and gave it another go. 

Great success! I have not seen the issue arise in any form since flashing them this last time almost a week ago.

TL;DR: You need to ensure a stable connection during the flashing of the firmware. If it comes loose a little bit, it might pass verification but still error on boot.

daredoes, 2020-12-01:
A secondary and maybe related issue I was having with other lights was that my 2.4GHz network only has a speed of 1mbps or less, which wasn't effective when transmitting data to my 5 WLED devices plus all the IoT I have as well.

Threw up a second local network with a faster 2.4GHz network, and the other issue disappeared, so maybe check your local wifi speed, not strength?

Irix0, 2021-01-15:
> A secondary and maybe related issue I was having with other lights was that my 2.4GHz network only has a speed of 1mbps or less, which wasn't effective when transmitting data to my 5 WLED devices [...]

I'm having the same issue, the problem doesn't come from network. The problem is quite strange, it appeared without updating or changing the settings, I tried different flash methods (ESPHome and esptools.py) (flashing via micro usb) and nothing worked.

I can give more details about the issue : 
The slightest small interaction with the web interface (only when ESP32 is connected to a network, it works fine AP) crashes (uptime to 0s) the ESP32.

Irix0, 2021-01-15:
I think it's related to #1431 and fixed with #1465

Irix0, 2021-01-16:
Fresh compile and flash, issue persists.

closed 2021-06-09 by stale[bot]