wled/WLED #1027: Ethernet support

aaronk6, 2020-07-05:
I would love to see Ethernet support in WLED to minimize the latency when syncing LED strips to audio.

There are two ESP32 boards that have Ethernet:

* Olimex ESP32-POE _(I have a few of these)_
* wESP32

Ethernet is working fine in ESPHome: https://esphome.io/components/ethernet.html However, ESPHome it is far less powerful compared to WLED when it comes to lighting and effects, so it’s not a real alternative for this particular use-case.

Related posts from the Discord forum:

* https://wled.discourse.group/t/is-it-possible-to-use-the-esp32s-ethernet-connection/475
* https://wled.discourse.group/t/wired-networking/567/2

Aircoookie, 2020-07-05:
I agree! Ethernet is also useful in areas where WiFi reception would be bad or would open up to using WLED in more professional applications. Ordered an Olimex POE board and will get to this soon!

AndrewApi, 2020-07-22:
How about using Adafruit Ethernet FeatherWing, it looks like it fits on top of a nodemcu?

Datasheet attached.

[adafruit-wiz5500-wiznet-ethernet-featherwing-1396496.pdf](https://github.com/Aircoookie/WLED/files/4958685/adafruit-wiz5500-wiznet-ethernet-featherwing-1396496.pdf)

dikt1338, 2020-10-26:
Hi @Aircoookie 
which Ethernet board did you order for testing?
Thanks

itavero, 2020-11-04:
I've ordered an ESP32-GATEWAY today, as I also prefer a wired connection over a wired one.
I would prefer to use the built-in Ethernet MAC over adding additional components.

tbnobody, 2020-11-04:
I applied several patches to the original WLED code for testing purposes... seems to work very nice: https://github.com/tbnobody/WLED/tree/ethernet

I am using the Olimex ESP32-POE Board: https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware

Legsmaniac, 2020-11-04:
> as I also prefer a wired connection over a wired one

Hmmmm, sounds intriguing. How does that one work? 😂

Aircoookie, 2020-11-04:
Also have the Olimex ESP32-POE(-EA)

@tbnobody that's awesome! If it doesn't cause issues for WiFi-only boards I'd be happy about a PR :)

tbnobody, 2020-11-04:
@Aircoookie to be honest I didn't know whats the best way to handle a board with wifi and eth capabilities. So I decided to disable the wifi interface when the eth interface is connected and retrieved an IP. But that also leads to the problem that every call of Wifi.Localip() (e.g. in the usermods or libraries) would return the wrong ip. So I decided to implement the Network class (https://github.com/tbnobody/WLED/blob/ethernet/wled00/Network.cpp) which returns the right IP/netmask/gateway etc. based on the current connection state. 
Don't know if this method suits you. 
I can send you a PR (after rebasing to your latest commit in the master repo) and we can discuss the required functionality there, or just here :-)
Currently I would say its a work in progress status.

itavero, 2020-11-05:
@Legsmaniac It was a long day, I guess. 😅

I'm not very familiar with the WLED code base, but I browsed through the changes by @tbnobody yesterday. The abstraction for the Network makes sense. I also think disabling WiFi when there is an Ethernet link makes sense.

I'll have a more thorough look later today and add my feedback to the PR (if I have any).
I think there will be a few interesting edge cases, like the Ethernet link is active but there is now DHCP server on the network and no static IP has been configured yet. In such a case WiFi should probably still be activated to do the configuration.

Unfortunately my ESP32 boards with Ethernet are still on their way from Texas to the Netherlands, so I can't test anything just yet.

aaronk6, 2020-11-07:
I’ve tried @tbnobody’s branch today. At first glance, this is working well and audio visualization via E131 is _much_ smoother now! 🥳

However, for some reason, the connectivity gets lost when powering off the LED strip through the WLED web interface (clicking the Power button in the upper left). Once the fade is complete, the green status LED on the board is turned off and network connectivity is gone. This is reproducible for me—video attached.

Any idea why this is? I’m also using the Olimex ESP32-POE.

---

This is what happens when I click the Power button in the app:

[image]

[image]

tbnobody, 2020-11-07:
I cannot test it at the moment because I used the ESP32-POE for a CO2 sensor project... BUT

in NpbWrapper.h
```
#ifndef RLYPIN
#define RLYPIN 12  //pin for relay, will be set HIGH if LEDs are on (-1 to disable). Also usable for standby leds, triggers,...
#endif
```

And in the file https://github.com/espressif/arduino-esp32/blob/master/variants/esp32-poe/pins_arduino.h
`#define ETH_PHY_POWER 12`

That means that the relay function of the WLED Source also powers off the ETH_PHY

itavero, 2020-11-07:
I'll try to do some testing next weekend. I expect to get my boards in on Monday.

aaronk6, 2020-11-08:
@tbnobody Awesome, that worked! I’ll monitor this over the next couple of days and report back if I encounter any other issues.

Aircoookie, 2020-11-12:
Tested out #1316 on my ESP32-POE-EA today!
Works very well in my testing and I'm really glad to have ethernet as an option especially for areas where a strong WiFi signal would be hard to achieve or even zones where WiFi is not allowed. It's blistering fast as well.

On an unrelated note, the external antenna on the -EA version of this board is great too. Without ethernet hooked up, I get at least 15dBm better signal than with the same distance and the regular PCB antennas!

itavero, 2020-11-13:
I've received an ESP32-GATEWAY-IND this week. Hopefully I'll find some time this weekend to also give it a go.

Aircoookie, 2020-11-20:
Closing since Ethernet support is now merged :) Feel free to use this thread though if there are any further findings or issues regarding Ethernet!

closed 2020-11-20 by Aircoookie

lonestriker, 2020-12-04:
WLED ethernet support is awesome, I was just going to look into it and you've already gotten POE ethernet boards integrated.

Is there any chance you can integrate support for the cheap W5500 boards?  No POE needed since we have to drive the LEDs with power already.  I've got ESP32 boards lying around that I would love to add ethernet to.

Here's the part on Amazon, you can obviously get them much cheaper from China:
https://www.amazon.com/gp/product/B07MYY7QW2/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Here's the wiring for the board to ESP32:
https://esp8266hints.wordpress.com/2019/04/05/esp32-w5500-simple-working-example/

This is the wiring of the board to ESP32:
https://github.com/jozala/ESP32_W5500_MQTT#physical-connection-between-esp32-and-w5500

GPIO23 <--> MOSI
GPIO19 <--> MISO
GPIO18 <--> SCLK
GPIO5 <--> SCS
GPIO11 <--> RESET

Aircoookie, 2020-12-04:
@lonestriker I agree W5500 support would be cool. It would also work with ESP8266. However, it does not use the TCP/IP stack of the ESP32 (like the POE board), but its own. That could be challenging, because we'd need to get all the Async network libraries to work well with the default Arduino Ethernet.h library. Not sure it is feasible, but if someone wants to give it a try this is certainly something I'd love having!

lonestriker, 2020-12-04:
Ah, I didn't realize we would have to port/integrate a TCP/IP stack as well to get the W5500s to work.  Definitely seems like a lot of work.  I will try to grab a couple ESP32-POE-EA adapters as well as some ESP32s with external antenna connections to see if I can get past my E1.31 latency issues (I'm running quite a few WLED and Tasmota devices.)  If I get a chance later, I'll have a look at what it would take to get TCP/IP on the W5500s, but won't happen until at least after Christmas.

Thanks for the quick reply and all your awesome work with WLED!

samstreak, 2020-12-07:
Just got my Olimex ESP32 Gateway Rev G board but nothing seemed to get the LEDs to light up. The only way I could get the Ethernet port to show link (and get the web interface) was to compile it myself as the ESP32-POE option (didn't try others, that was my first one and it worked immediately for the Ethernet), and I tried several of the GPIO's for the LED pin but none seemed to do anything. What GPIO's did you all use to test with?

lonestriker, 2020-12-07:
I got the same one as Aircookie, the ESP32-POE-EA (comes with external wireless antenna as well.)  I compiled it as well and am successfully using the default LEDPIN=2.  WLED is amazing on ethernet.  It's as performant as dedicated E1.31 devices like a Falcon F16V3 controller.  If my wireless weren't so bad (I've tried two different mesh type systems a Ubiquiti and TP-Link Deco and it's still laggy).

CaptInsano, 2020-12-07:
I just ordered a ESP32-POE-EA after seeing ESP32 Ethernet support on 0.11 release notes.

I did not see this question answered (sorry if I missed it somewhere): 

Is support provided in the WLED_0.11.0_ESP32.bin or must we compile a build for ethernet support?

Congrats on the amazing work!

tbnobody, 2020-12-07:
Imho its currently not included in  any released binary. So you have to compile it yourself. 
The flag -DWLED_USE_ETHERNET is only set at the esp32_poe environment in platformio.ini (also the correct board type which is required to setup the right pins for the ethernet controller)

swinbkh, 2020-12-08:
I've ordered both an Olimex ESP32-POE, and a WT32-ETH01 board to test ethernet functionality.

There is very little docuementation for the WT32-ETH01, but it is 4x less expensive. It has been suggested by arendst, developer of Tasmota that both boards should work without firmware modifications. I'll be sure to provide an update when they get here!

Aircoookie, 2020-12-09:
WT32-ETH01 indeed seems like great value! Will be a pain to flash initially for the lack of USB or at least an FTDI header, but once WLED is on it, it seems like a nice choice to be able to use Ethernet at pretty much no extra cost (in comparison to a "regular" ESP32 devboard)

lonestriker, 2020-12-09:
Agreed, thanks much for the pointer @swinbkh!  Don't need POE so the $20 savings per board is enough to buy almost 4 more.  My order of 5 has shipped and looking forward to running WLED on it.  Just have to pre-flash WLED on all of them and use OTA for updates later.

m-s-e, 2020-12-16:
> Will be a pain to flash initially

Hay all,

I'm not sure how we would go about flashing a WT32-ETH01 if it's not posible to connect via USB. I'm new to this kind of tech, so have tried to educate myself a tiny bit...

 [on this page](https://www.makeuseof.com/this-6-esp32-board-is-a-diy-smart-home-dream/) it says:

```
It lacks an onboard programmer, though it can be programmed via any generic USB to TTL converter, much like the Arduino Pro Mini. Another notable omission is Power over Ethernet (PoE), which would've been convenient. Given the $6 price tag, making this board cheaper than any other with an Ethernet connector, it's likely not too big of an issue.

The board comes flashed with firmware that uses the AT command set (also known as the Hayes command set), though it works with Tasmota, an open-source software build supporting a large range of smart home devices.
```

I'm sure that means a lot more to most people in this thread, than it does to me!!

""

lonestriker, 2020-12-17:
@m-s-e It just means you have to buy an FTDI USB-to-serial device (something like [this one](https://www.amazon.com/HiLetgo-FT232RL-Converter-Adapter-Breakout/dp/B00IJXZQ7C/ref=sxts_sxwds-bia-wc-p13n1_0?cv_ct_cx=ftdi&dchild=1&keywords=ftdi&pd_rd_i=B00IJXZQ7C&pd_rd_r=fd97fcab-d138-4332-9c47-bf1560fd9463&pd_rd_w=LZSlE&pd_rd_wg=0s6gA&pf_rd_p=1835a2a9-7ed8-48dc-ad07-fcd7527bd2bc&pf_rd_r=Q4JX3FBKZ2T82FFYMPG9&psc=1&qid=1608239323&sr=1-1-80ba0e26-a1cd-4e7b-87a0-a2ffae3a273c)) and some jumper wires to connect the FTDI device to the ESP32-ETH01 device so you can flash it.  Basically, it's an external USB device that talks to the ESP32-ETH01 board to program it.

m-s-e, 2020-12-18:
Great, thanks. I should have a [USB > TTL cable somewhere, like this one](https://www.amazon.co.uk/dp/B00514ZCHQ/ref=pe_385721_37986871_TE_item)
somewhere.
I guess that'll work if i get the pip links right?

lonestriker, 2020-12-19:
I've never used that one before, but if it has an FTDI driver that's recognized by Windows/Mac/Linux you should be good. There may be a possibility that the cheap ethernet boards come with a firmware that allows OTA flashing. I'll see if I can find Wi-Fi or ethernet connectivity to the default firmware before I flash using FTDI interface.  My boards are in the US at this point, so hopefully delivery some time next week.

m-s-e, 2020-12-21:
Great. I look forward to hearing about your experience!...

swinbkh, 2020-12-21:
> > Will be a pain to flash initially

I agree, but once flashed it should always be possible to upload newer WLED revisions through the OTA utility built in. For the initial upload, something will be required. Typically, as @lonestriker suggested, you'll be able to use an arduino as ISP or FTDI board. It's a fairly simple upload process, and fairly hard to break your boards, even if it goes wrong and takes a few times. 

I can't imagine the lack of PoE will be an issue, since for most WLED applications, there is an external power source, which can be used to provide power to the board as well as the lights. PoE seems like an unnecessary additional expense that won't be utilised.

lonestriker, 2020-12-23:
OK, so I got three of the boards delivered today

TL;DR: they work!  But with some code changes needed.

Some notes:

* I couldn't figure out a way to flash using the existing firmware (I didn't really try too hard other than looking for a Wi-Fi AP or plugging into ethernet and hoping for a new DHCP address showing up on my network.)
* I used a standard FTDI adapter to flash new firmware on these boards.  You have to jumper GPIO-0 (zero) with ground to program since there's no button to do that on these devices.
* Use TX0 and RX0 pins and **NOT** TXD and RXD ones to connect to FTDI.
* I used 5V on my FTDI to program and then run the board to watch logs (3.3V isn't enough to power this WT32-ETH01 board and I got brownout errors.)
* I verified that Tasmota works with it compiling a version with USE_ETHERNET defined and uncommenting out the my_user_config.h ETH_TYPE, ETH_ADDR and ETH_CLKMODE settings for the WT32-ETH01
* You also have to configure the [WT32-ETH01 Tasmota template](https://tasmota.github.io/docs/ESP32/#wt32-eth01) as well before the ethernet adapter works.
* I was able to take the Tasmota settings and hack them into WLED and can use the WT32-ETH01 just the same as the ESP32-POE adapter (minus of course the POE power injection.)
* I'm able to use this board + modified WLED to run as part of my E1.31 Christmas lights display.

So definitely grab some of these boards if you need ethernet for your WLED runs.  Particularly useful for lag-free E1.31 xLights displays.  I'll clean up the code and submit a PR after some more testing.

jonslilbro, 2020-12-23:
> OK, so I got three of the boards delivered today
> 
> TL;DR: they work! But with some code changes needed.
@lonestriker  care to post a little tutorial or the firmware for it? I was able to get WLED 011.1 onto mine but haven't been successful getting ethernet to work at all(no blinking lights on the port even). Would really love to get this one going. Thanks for your work!

m-s-e, 2020-12-23:
this all sounds really exciting!! i'm glad the board is a success!

lonestriker, 2020-12-23:
Fork, source changes and compiled release here:
https://github.com/lonestriker/WLED/releases/tag/v0.11.1-wt32-beta

@jonslilbro just OTA flash the firmware or FTDI if you can't access via Wi-Fi.  Let me know if it works for you.

jonslilbro, 2020-12-23:
@lonestriker that did it! Thanks!

swinbkh, 2020-12-23:
@lonestriker, brilliant work. I'm eager to get it loaded and tested when mine arrive, but great work adding support so quickly. I recognised bits from tasmota, and would have done the same thing to get it up and running. ;)

With multiple ethernet boards supported now, it may be worth creating a Wiki page just to document how to use it once your PR is approved, just for things like initial configuration etc. I'd be happy to get started on that!

pk88888, 2020-12-23:
hi! is it possible to set a static ip address for ethernet connection? 
Now if i connect via wifi it has 192.168.1.116 and 192.168.1.117 when using ethernet.

tbnobody, 2020-12-23:
> Fork, source changes and compiled release here:
> https://github.com/lonestriker/WLED/releases/tag/v0.11.1-wt32-beta
> 
> @jonslilbro just OTA flash the firmware or FTDI if you can't access via Wi-Fi. Let me know if it works for you.

I don't have such a board yet, but I think you can undo your changes in wled.cpp and just add the following to the build flags:
```
-D ETH_PHY_ADDR = 1
-D ETH_PHY_POWER = 16
-D ETH_PHY_MDC = 23
-D ETH_PHY_MDIO = 18
-D ETH_PHY_TYPE = 0
-D ETH_CLK_MODE = 0
```

m-s-e, 2020-12-24:
@lonestriker are you using IO2 for the signal to the LEDs?

lonestriker, 2020-12-24:
> > Fork, source changes and compiled release here:
> > https://github.com/lonestriker/WLED/releases/tag/v0.11.1-wt32-beta
> > @jonslilbro just OTA flash the firmware or FTDI if you can't access via Wi-Fi. Let me know if it works for you.
> 
> I don't have such a board yet, but I think you can undo your changes in wled.cpp and just add the following to the build flags in the platformio.ini file:
> 
> ```
> -D ETH_PHY_ADDR = 1
> -D ETH_PHY_POWER = 16
> -D ETH_PHY_MDC = 23
> -D ETH_PHY_MDIO = 18
> -D ETH_PHY_TYPE = 0
> -D ETH_CLK_MODE = 0
> ```

Thanks for the suggestion. I'd rather keep the settings as a run-time setting rather than compile-time.  At some point, the type of Ethernet board used could be a WLED configuration option instead of needing a special build just for the WT32-ETH01 vs. ESP32-POE vs. some other variant.  It's hard-coded now in wled.cpp, but shouldn't take much to add a configuration option to toggle and read from that instead.

@m-s-e yes, I left the default pin for WLED which is GPIO2.

lonestriker, 2020-12-24:
> @lonestriker, brilliant work. I'm eager to get it loaded and tested when mine arrive, but great work adding support so quickly. I recognised bits from tasmota, and would have done the same thing to [...]

I generally look for the quickest way to hack in fixes to get them working.  So the least amount of effort was to find something that worked (i.e Tasmota), add some debugging to see what settings they used, and then just inject those into WLED.  I'm not terribly well versed in either Tasmota or WLED code, but I can search/grep and cut-and-paste with the best of them :-)

Would be a good idea to document the idiosyncrasies of each board (how to program, what GPIO pins work and which ones don't etc.)  If you're volunteering, I'm all for it. Documentation is not my strong suit.

lonestriker, 2020-12-26:
> Thanks for the suggestion. I'd rather keep the settings as a run-time setting rather than compile-time. At some point, the type of Ethernet board used could be a WLED configuration option instead of [...]

Spent a few hours to hack in the configurable ethernet board support.  That took a bit longer than I expected, mostly just dealing with the web stuff.  You can get the firmware and code changes [at my fork](https://github.com/lonestriker/WLED/releases/tag/v0.11.1-eth-beta-2).  One of the benefits of this change is that we should be able to enable ethernet support as part of the base ESP32 binary and not need a separate one for Wi-Fi-only vs. ethernet.  It only activates when you configure an ethernet board (defaults to "None".)  Just connect to the WLED-AP and instead of setting up your Wi-Fi connection, just select your ethernet board at the bottom of the settings page.

@tbnobody this is what I had in mind, would love to hear your thoughts.  And thanks much to you and @Aircoookie for the amazing software and features.

tbnobody, 2020-12-26:
I think it make sense to have this configuration as a runtime option as it's easier for many people to use a pre-compiled binary.

Just a hint, based on that file: https://github.com/espressif/arduino-esp32/blob/master/variants/esp32-poe/pins_arduino.h
the eth_power for the esp-poe board should be 12

lonestriker, 2020-12-26:
I just found the power bug as well when trying the binary on my ESP32-POE board.  Thanks for the pin corrections, should have verified it worked on both before publishing.  Thanks again!

k7bbr, 2020-12-29:
I just received my WT32-ETH01 boards from Seeed Studio. I was able to flash @lonestriker compiled version to the boards with no issues. I'll do some testing and report any issues. I like the option to select the ethernet variant via the initial wifi page. I may recompile mine to use a relay pin (not pin 12) however, to interface it with a multiplexer board I'm working on. Thanks!

lonestriker, 2020-12-29:
> I just received my WT32-ETH01 boards from Seeed Studio. I was able to flash @lonestriker compiled version to the boards with no issues. I'll do some testing and report any issues. I like the option [...]

Glad to hear it's working out. I've always wondered if making the output pin also configurable via the web menu (like I did here with the ethernet none, poe, wt32 option) would be feasible. Would save us from having to compile different versions with just an output pin difference.

k7bbr, 2020-12-29:
> Glad to hear it's working out. I've always wondered if making the output pin also configurable via the web menu (like I did here with the ethernet none, poe, wt32 option) would be feasible. Would save us from having to compile different versions with just an output pin difference.
>
I like the idea of a configurable output pin via the web menu. Would be nice to continue the theme of a generic Ethernet board version that is user configurable for ethernet board type as well as relay pin. Would want to make sure pin options were only those that were feasible/available outputs and wouldn't bring up conflicts with other pins.

k7bbr, 2021-01-01:
So I've compiled to try various relay pins. Right now I've turned off IRPIN and made RLYPIN 4 for now. Ethernet seems to be working. However, I do get random instances where WLED turns the lights off (as if I'd pressed the power button in the web interface or on the app). Everything else works. When the power button appears to randomly get turned off, I can go back in and turn it back on. 

I briefly tried a version of the master branch without ethernet support for the WT32_ETH01 board. It seemed to work without the random turning off of the lights. Don't know if you want me to keep the discussion going here, or create an issue on @lonestriker 's branch to figure out what's happening.
Thanks for everyone's work on the project and on this Ethernet board. I can't wait to swap out some of my wireless ones.

swinbkh, 2021-01-01:
My board is also randomly powering on/off LEDs even with a fresh, unmodified version of @lonestriker fork, which can be overridden with a press of the power button within the web page. Left to its own devices, it continually powers on/off the LEDs. I think this suggests it's unlikely to be a one off!

m-s-e, 2021-01-01:
I'm also experiencing the same thing (random LEDs on / off) with the fork from @lonestriker .
The board runs quite hot (compared to a D1mini), could it me an overheating issue?
I'm using a 5V PSU. Are others using the 3v3 input?

k7bbr, 2021-01-02:
Glad I'm not the only one seeing the on/off issue with the @lonestriker fork. I have tried multiple pin assignments the last couple days. I've successfully ran my WT32-ETH01 board for over 24 hours without seeing the power on/off issue. The config I'm using takes the @lonestriker beta 3 platform.ini for the esp32-eth environment and changes:

-D RLYPIN=4
-D IRPIN=-1

This disables the IRPIN, which for some reason seems to solve the random on/off issue.

pk88888, 2021-01-02:
Are you sure it's a power problem? Recently, I noticed a problem with devices resetting after turning on Advanced IP scanner (https://www.advanced-ip-scanner.com/). When the program finds a given device and shows it on the list, the specific WLED device is reset.

Resetting also happens in the normal firmware version 0.11.1

I haven't checked what commands this software sends yet, but maybe it's the same problem with all of us.

swinbkh, 2021-01-02:
@pk88888 , I don't think it's caused by low power, I was more describing the symptom- the "UI power" button is effectively pressed, turning the lights on and off, but the board remains online and working. 

I have noticed my WT32 board runs hot too, but again, this could just be a symptom rather than the cause, as the "power" issue can start before the board has warmed up.

I'll change my pin assignments and see if the issue persists

lonestriker, 2021-01-05:
I've been offline for a bit for New Year's so didn't see this traffic.  

I have noticed some instability recently as well with the latest build and running an xLights E1.31 show.  My LEDs randomly stopped displaying lights (as though the power button was pressed) but the WT32-ETH01 is still running and says it's under E1.31 control.  I originally attributed this to my changes to adding more lights, but with your reports, it may be a software issue.

Are you guys seeing the issues with all of my BETA builds?  The first two builds were from the 0.11.1 base codebase.  The BETA 3 build included the latest commits from the official WLED github merged in.

My changes really are not very invasive, at their core, they basically just take the ESP32-POE support and initialize the board with different pins specific to the WT32-ETH01 board.  

A good test to see what the issue could be would be to take the stock WLED build and just compile in the pin assignment changes as @tbnobody suggested upthread:

> I don't have such a board yet, but I think you can undo your changes in wled.cpp and just add the following to the build flags in the platformio.ini file:
>
> ```
> -D ETH_PHY_ADDR = 1
> -D ETH_PHY_POWER = [...]

I'll see if I can generate a simple build with stock WLED and the compiler flags above and see if I still see the same stop issues.  I'll post when there's a bin to test.

k7bbr, 2021-01-05:
I'll give that a try and let you know. 

I did try to compile as you suggested from Aircookie's master branch but ran into issues with build flags not being correct. 

As a side note, my build from your beta-3 branch is still running without random power offs (going on about 3 days now). My change on that one was to turn off the IRPIN and make RLYPIN=4 .

lonestriker, 2021-01-06:
I've been doing a bit of testing as well.  I'm not sure if any of the 0.11.1-based (and higher) builds I have are entirely stable.  I have a separate 0.11.0 build that's based on the multi-pin mod by [PeacePenguin](https://github.com/peacepenguin/WLED) that seems to be much more stable.  Simple things like just booting up and getting the ethernet connection initialized and link established doesn't work 100% of the time for 0.11.1+.  But I don't seem to have that issue with the 0.11.0 PeacePenguin-based build.   So perhaps there's something in the commits after 0.11.0 that changes timings or initialization that causes issues.

I'll keep testing and get back to you guys with my findings.  If you can let me know if you find any common or reproducible issues, hopefully we can narrow the cause down.

k7bbr, 2021-01-06:
I managed to sort out build flags and compiled the latest 0.11.1 version from Aircookie's master branch. It's running now. I'll keep an eye on it and see how stable it runs. Maybe later I'll try it on a 300 pixel prop I just took down from my holiday display. Interested to see how E1.31 performance is vis Ethernet.

lonestriker, 2021-01-06:
@k7bbr I'm trying a clean 0.11.0 build now with just the compiler flag changes for WT32-ETH01 board to see if it's any better than 0.11.1+ with the compiler flag changes only.  

As for WLED E1.31 performance over ethernet, it's completely flawless.  I run both a Falcon F16v3 as well as WLED and the performance between the two is indistinguishable.  No lags, no delays.

[33 more replies]