wled/WLED #1028: Wled Alexa Sonos

bolek1, 2020-07-05:
Alexa App findet keine Geräte.
Aktuelle Wled Version.
Ist da was bekannt oder funktioniert es nur mit original Alexa Device?

Aircoookie, 2020-07-05:
Hi, also mit den original Echos funktioniert es meist gut, mit dem Fire TV oder nur der Alexa app allerdings nicht. Ich fürchte der Sonos fällt unter die zweite Kategorie. Sobald ein Echo Gerät im Netzwerk ist, lässt sich WLED aber auch über die von sich aus nicht kompatiblen Geräte steuern. Die dafür verwendete lokale Hue API is offenbar nur bei den Echos implementiert, die dann als Gateway dienen.

odins-peritas, 2020-09-03:
Summing up in english:
Yes, it only works, if a Amazon Alexa Hardware with inbuilt bridge is in the network. Or a Philips Hue bridge. You can see this, if you go to your Alexa app (e.g. with only a Sonos working with alexa) and try to add a hue lamp. it will ask for a echo dot 3rd generation, echo plus or hue bridge (list not complete)

closed 2021-01-09 by stale[bot]