wled/WLED #1030: Exception 28 after configuring wireless settings

gblanchard, 2020-07-06:
After configuring wireless settings and rebooting for the first time ESP01 crashes almost immediately

WLED_0.10.0_ESP8266_1M_full.bin

```
Exception (28):
epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd20 end: 3fffffc0 offset: 01a0
3ffffec0:  00000000 faffffef 80000000 3fff1c34  
3ffffed0:  3fffff30 3fffff3c 00000000 4022b297  
3ffffee0:  00000050 3fffff10 3fff17c0 3fff19ff  
3ffffef0:  00000003 3fff1be0 3fff10d0 3fffff3c  
3fffff00:  3fffff30 00000050 3fff1c34 4022b302  
3fffff10:  3ffea507 3fffff30 3ffea503 3fff19ff  
3fffff20:  3ffea507 3ffea503 3fff1c34 4021cf34  
3fffff30:  00706374 6264a8c0 83c6a7f0 64656c77  
3fffff40:  3fff5200 84223838 40100ae4 3fff1e2c  
3fffff50:  3fff1a05 00000000 3fff0ce0 4021d1b2  
3fffff60:  00000000 48fc7c05 3fff0c00 3fff1e2c  
3fffff70:  00000000 00000000 3fff0ce0 3fff1e2c  
3fffff80:  3fffdad0 00000000 3fff0ce0 4021d221  
3fffff90:  3fffdad0 00000000 3fff1dec 4021db4c  
3fffffa0:  feefeffe feefeffe feefeffe 4023209c  
3fffffb0:  feefeffe feefeffe 3ffe8800 40100851  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
```

Aircoookie, 2020-07-07:
Hi and sorry for the issue!
Please erase the flash completely (for example with `esptool.py erase_flash`) and flash again.
Once the firmware is running again, before entering your home network information, please go to Sync settings and disable "Emulate Alexa device". In Wifi settings, clear the mDNS fields.
This problem seems to be widespread and limited to the `1M_full` binary. Need to investigate :)
Likely same cause as #1025

gblanchard, 2020-07-09:
I erased the flash and still same issue. Haven't disabled sync yet though.

gblanchard, 2020-08-31:
No worries, will try again when I get the chance. Thanks for following up.

sector7gp, 2020-09-22:
It works for me! I got some olds ESP8266EX 512K

Thanks!

closed 2021-01-31 by stale[bot]