wled/WLED #1032: Safe mode recovery

Aircoookie, 2020-07-07:
**Is your feature request related to a problem? Please describe.**
Bootloops require complete reset of the flash contents, erasing all user settings and being a major inconvenience for controllers in hard to access areas.

**Describe the solution you'd like**
A safe mode that boots WLED with :
- reduced LED count
- AP mode only
This should either be activated after a set number of unsuccessful boots (would require writing a flag to flash at every boot, not ideal for flash longevity, but can be manageable with the planned introduction of a wear levelling filesystem).
An alternative would be checking the reset reason flag and loading safe mode if the reset was not either planned or a cold start.

HHHartmann, 2020-07-08:
Being new to this project (but not to ESP8266) may I suggest storing the reboot count in the RTC memory? 
This works great for me.

0rsa, 2021-05-05:
+1, I just brick my Mini D1 pro updating to 12.0.0 and I have no physical access to the ESP8266. I learned a lesson :=)

blazoncek, 2023-01-25:
`power-ap` branch (with power-ap usermod) features forced AP mode on 3 power cycles.
May be implemented as a core feature to either force AP mode and/or boot in safe mode.

DedeHai, 2025-08-31:
https://github.com/wled/WLED/pull/4793 reverts the config and even rolls-back a failed OTA. However, if there is a boot-loader mismatch, it wont help but it is probably as good as it gets.