wled/WLED #1033: WLED submits invalid DNS hostnames to DHCP

garyd9, 2020-07-08:
**Describe the bug**

WLED appears to submit the "Server description" text field as a hostname when registering via DHCP.  The "server description" often will contain characters that are invalid machine names and if the DHCP entry is passed along to DNS, a DNS server can't properly handle the submitted server name.

**To Reproduce**

In a new WLED instance running on a wifi network that uses DHCP and where DHCP forwards names to DNS, enter config, select "User Interface" -> and type a "Server description" that contains spaces.  For example, "Front Porch Lights."  Save.  Go to WiFi Setup and ensure no static IP information is provided (or that it's all 0.0.0.0.)

**Expected behavior**

WLED would submit a hostname with DHCP that meets the requirements of RFC 952 and 1123.   (from 1 to 63 characters long, letters from a to z, the digits from 0 to 9, and the hyphen (-).  A hostname may not start with a hyphen)

One alternative might be to replace invalid characters in the "server description" with hyphens.  Another might be to use a configured mDNS name.  As a fallback, use the MAC address.

**WLED version**

 - Board: esp8266
 - Version: 0.10.0
 - Format:  binary from WLED site

**Additional context**

I validated the issue using Windows Server 2016 DHCP that forwards to Windows Server 2016 DNS (which drops the DNS entry that DHCP attempts to create and logs an error.)

garyd9, 2020-07-08:
I'd love to propose a fix, but I'm not familiar with what libs are available for use.  It appears that std c++ strings are, however.  I just don't know what version of C++...  So, perhaps in wled.cpp, ~line326 (before the #ifdef ESP8266):

```
// NOTE:  this is off the top of my head and should most certainly be tested if it even compiles.

// mangle the serverDescription to be a valid DNS hostname by replacing invalid chars with hyphens.
String hostname(serverDescription);
for (auto it = hostname.begin(); it != hostname.end(); ++it)
    if (!::isalnum(*it))  *it = '-';   // replace any non alpha-numerics with a hyphen.. yes, a hyphen replaces itself
// ensure the first character isn't a hyphen
if ((0 < hostname.size()) && ('-' == hostname[0]))  hostname.erase(0,1);

```
then, there are two lines of code that follow that called "WiFi.hostname(serverDescription)".  Just change them to pass in hostname instead.

garyd9, 2020-07-08:
I was way off on the string type.  Wow.   (I'm doing too many different platforms at one time in my daytime/paying job.)  Even still, the idea is still the same and still trivial: iterate through the string replacing non alphanum with hypens.

garyd9, 2020-07-09:
I'm not sure how to label a pull request to mark it as a potential resolution for this issue... I just sent pull 1035 that would resolve this.

Aircoookie, 2020-07-10:
Thank you for noticing this, I was not aware that hostnames are only spec'd to contain alphanumeric characters.
Will merge your PR soon!

closed 2020-07-19 by Aircoookie